jotego/jtcores #248: Super Street Fighter 2 Turbo Graphical issue when playing Akuma

wereoffs, 2023-04-21:
When selecting Akuma as a player in Super Street fighter 2 Turbo, there is a black box appearing on the left of the screen. 
(I don't know if it is a coincidence, but a similar bug happens randomly, but rendering random sprites in the same zone, this time when playing with Ryu.)
[image]

Reproductivity: 5/5

Steps to reproduce:
Launch Game
Insert coin
Press start (player 1)
Select/unlock Akuma character (using the right code sequence) ([https://twitter.com/victoralonso_es/status/1647729867755073536](https://twitter.com/victoralonso_es/status/1647729867755073536))

The black box appears during the fight gameplay,Win / lose /map scenes looks ok.

wereoffs, 2023-04-21:
Oh, just found in the linked video that it happens as well in the original hardware, so not a core issue. sorry :D

closed 2023-04-21 by wereoffs

jotego, 2023-04-21:
Thank you for investigating this.