jotego/jtcores #257: X-arcade Tankstick Direct and Xinput not supported

wiggy808, 2023-05-02:
When prompted to register the buttons in the core for the second joystick, nothing registers when moving the directions of the second joystick. 

This occurs for both dinput and xinput modes on the xarcade tankstick. 

Possible solution is to add a core option like what’s in Robotron 2084, where it has the core option “Fire” which lets you choose:
- 4 way
- Move with fire 
- Second joystick

The option “second joystick” in that core allows the second joystick on the xarcade tankstick to be used when in dinput or xinput modes, however no such option exist and current mapping prompts do not allow second joystick inputs on the xarcade tankstick for xinput or dinput modes.

jotego, 2024-05-05:
Related to #649

jotego, 2024-09-20:
I think this was fixed when #649 was fixed.

closed 2024-09-20 by jotego