jotego/jtcores #262: the final round: dump 4-bit ROM

jotego, 2023-05-10:
This device has no known dump:

[image]

etriana85, 2023-05-15:
Good. I'll work on this.

etriana85, 2023-05-25:
Best regards team!
Checking the inventory of the pcb's, I realize that we don't have the final round. What we have of this game is the manual and the schematic.
On the other hand I checked the pcb Gradius II ( Devil Word), but they don't use the 63S141 memory. 
Konami GX, we don't have the pcb, but in the list of known games Konami Twin 16, I see that there is M.I.A, and when I check it I detect that this pcb uses the 63S141 memory.
According to this I remain attentive to new instructions, since this pcb is not ours.

etriana85, 2023-05-25:
[image]

jotego, 2023-05-25:
Because this is a PROM device, the contents will likely be different for this board so it isn't good for us. Maybe @skutis can comment too.

skutis, 2023-05-25:
Ok, I see. I thought you had the final round. I don't think it will be different, however. But we can not be sure. I can desolder it from my spare the final round and send it to you.

jotego, 2023-05-25:
There's a good chance it will be different. That kind of device is common on many boards. We can dump it if you send it over and send it back.

atrac17, 2023-05-25:
Socket the chip off of MIA and pull it off and dump it. MIA is mine, I don't have an issue with that.

jotego, 2023-05-26:
Thanks, @atrac17. But the one from MIA is already available:

```
ROM_REGION( 0x0100, "proms", 0 )
ROM_LOAD( "808a18.f16",   0x0000, 0x0100, CRC(eb95aede) SHA1(8153eb516ae9753910c6d6a2143e91e079586836) )    /* priority encoder (not used) */
```
I think the one in Final Round will have different contents.

etriana85, 2023-06-15:
Skutis has sent us;
- The device 63S141 of the final round to make the dump.
- The pcb "the final round", for the cores and/or measurements required.
The 63S141 device was unsoldered from the received pcb and a socket was soldered. I share images.

etriana85, 2023-06-15:
[image]
[image]

etriana85, 2023-06-15:
Between today and tomorrow we will dump this device and share the extracted information.

etriana85, 2023-08-28:
Reading extracted from PROM 63S141N.