jotego/jtcores #263: Alien Syndrome all versions isue, player freezes

tronerrr, 2023-05-10:
When playing, the player freezes and does not rotate. 
He can still move and shoot in all directions but he is a fixed figure that slides across the screen

christ776, 2023-06-17:
Hey, yeah , I'm having this very same issue as well. After a couple of minutes the player sprite freezes and it happens on all versions of the game. The sprite goes back to normal when loosing a life but it freezes again after a few seconds

florgana, 2023-06-23:
Bug fixed!
Bug happened with the keyboard, when 2 arrow key were pressed at the same time (example: left, right).

closed 2023-06-23 by florgana

christ776, 2023-06-23:
That's very interesting, because I was able to trigger this bug using my 8bitDo Arcade Stick. I'm sure there's a bug in the controller, because some time ago, I was having issues with Truxton II and the developer fixed it by also preventing left and right be pressed both at the same time.

atrac17, 2023-06-24:
Other "developer" here. I don't think it's the 8bitdo stick. I think it has to do with input handling in jtframe.

jotego, 2023-06-24:
I think it is the controller. JTFRAME could filter the joystick data and that's what we have added. But it is the controller who sent two contradicting signals in the first place. 

I think that controller produces glitches when you change direction.