jotego/jtcores #264: jtoutrun core broke system stability, analogue pocket issue

SkyNIC, 2023-05-10:
When I run jotego.jtoutrun on last analogue pocket firmware v.1.1 it looks like this

https://github.com/jotego/jtcores/assets/1808805/7de79b78-9737-492a-94d0-c6d0dd434adf

If I close jtoutrun core and run another core after for example spiritualized1997.genesis core and load contra hard corps game it will freeze in game or demo play scene.

After power off and power on spiritualized1997.genesis core loads without issues, but if I only load jotego.jtoutrun system stability broke until reboot

I tried previous firmware and got same issue, also another sd card with clean file sturcture 

md5 files
F6D3DC96523157A8D664A5B7FFE4E3F8 outrun.rom
60B9C5E6F27BF304708B1546E78193D9 toutrun.rom
A42F546D40C6B3205912B9CA82D8AB4C jtoutrun.rbf_r
2BD30479556DA5C310D8F2A1FE66E8FA Contra - Hard Corps (U) [!].bin

How to reproduce
Load jotego.jtoutrun with outrun.rom or toutrun.rom then load spiritualized1997.genesis core with contra hard corps game

SkyNIC, 2023-05-11:
I can reproduce stability issue not only with genesis core after running jtoutrun core. I can run jtoutrun after power on and then run Mazamars312.Neogeo core and it will boot with error message screen every time. I ran every installed core except jtoutrun then run Mazamars312.Neogeo it boot game without errors, but when I load jtoutrun and then load Mazamars312.Neogeo it boot game with error every time. 

I made video with issue, please watch

https://github.com/jotego/jtcores/assets/1808805/1548363b-106a-4735-a65d-eeb6d028ca38

I understand that you are not developer genesis or neogeo cores, but why only loading jtoutrun core break stability of other cores? Issues with other cores happens after loading jtoutrun

jotego, 2023-05-11:
JTOUTRUN uses the external cell RAM. Maybe those two cores use it too. The cell RAM has several operation modes and the core must set it up properly upon usage. If they don't set it up but just expect the default operation mode, then it will fail to work if the previous core changed it.

I think it's up to them to fix that.

SkyNIC, 2023-07-05:
[image]

jotego, 2023-07-05:
I think Kevtris is on the right track. If a core does not boot up, it is because it is not going through the right initialization sequence.

closed 2023-07-05 by jotego