jotego/jtcores #265: Cropped screen in jtdd and jtdd2 on analogue pocket

SkyNIC, 2023-05-11:
On analogue pocket display game looks cropped 
[image]
[image]
But if I make screenshot it looks fine on it
[image]
[image]

jotego, 2023-07-01:
I don't know what caused this but when tested on git commit 67256624, everything seems fine. I'll update the binary files.

SkyNIC, 2023-07-05:
Issue still present on latest core versions. Only on Analogue Pocket screen. When it connected through the Dock or screenshot looks fine. 
jtdd.rbf_r (14C09E98)
ddragon.rom (3E995248)
jtdd2.rbf_r (9578E5EA)
ddragon2.rom (83D0E4B1)
jtkunio.rbf_r (0AC9BFA9)
kuniounb.rom (C4C204EF)
renegdeb.rom (2440E583)

closed 2023-09-25 by SkyNIC