jotego/jtcores #267: Aliens: Random sprite corruption

djhardrich, 2023-05-13:
Video attached, look closely at the area around the Alien while Ripley is recovering.

Steps to reproduce: At the end of level one, get hit on purpose/fire while flashing after being hit. Corruption/displaying the wrong sprite generally manifests as a slice of the Alien's neck/head floating in space, Ripley's right or left side being replaced with the Alien's arms/legs every third frame (normal, alien, blank, repeat), or sprite priority from the gunfire displaying on top of Ripley while she flashes.

https://github.com/jotego/jtcores/assets/22062445/4ab7d5b2-3e84-4298-ae00-1ad96db44276

djhardrich, 2023-05-16:
Just tested in the new build (1.2.24), First Boss seems stable! Unfortunately, it crops up with Evil Newt and Alien-Bat Ripley in the subsequent stage if you get hit…

Evil Newt (Bottom Right):
[image]

Alien-Bat Ripley:
[image]

jotego, 2023-05-16:
Blast. No easy victory. That didn't happen before, did it?

djhardrich, 2023-05-16:
Not to my knowledge (played that stage 3 times on previous build and didn’t see it); there’s also something a little off with scrolling on a crt; it’s as if there’s a 15% frameskip/slight judder happening, could possibly be related? I’ll open a separate issue with more info/relevant video captures; Aliens might be the only game I’ve spent more time playing in the arcade than Street Fighter II!

jotego, 2023-05-17:
It looks like the status bit read by the CPU was inverted. Thus, the CPU was writting sprite data during blanking instead of during active video. Everything should work now well. I'm updating the files in a few minutes.

closed 2023-05-17 by jotego