jotego/jtcores #272: Service Menu / ROM Checks Fail / Grid Alignment [JTALIENS v1.2.29]

atrac17, 2023-06-10:
Thanks for working on Super Contra! Here's a few things I noticed after firing it up. ROM checks fail and service menu cannot be accessed by pressing 1P Start. Also, the reset line is still finnicky and can result in a red screen when attempting to access the service menu.

**[JTALIENS v1.2.29 Core Footage](https://youtu.be/z6-1iv2pE-8)**

**[MAME 0254 Footage](https://youtu.be/kN0i64Wokjs)**

Video grid isn't centered, a few rows of pixels are cut off (below). PCM audio doesn't always play, some sounds are muted in the mix. I'm unable to access the sound test to pursue this any further because you cannot cycle thru options after enabling service mode.

If I notice any other issues, I'll report them.
[image]

jotego, 2023-06-11:
The logic needed to check graphic ROMs is not implemented yet. It's the same problem in Aliens. That has low priority for now. The service menu will show bad ROMs for a while.

I fixed the PCM sounds problem. Thank you!

jotego, 2024-04-02:
Action on ROM checks moved to #557

closed 2024-04-02 by jotego