jotego/jtcores #273: aliens: scroll glitch?

jotego, 2023-06-11:
The horizontal scroll feels a bit glitchy. Running a simulation to sweep the horizontal scroll in scontra resulted in a smooth result. This needs one more check in an Aliens gameplay simulation to check whether h. scroll settings are modified mid frame or any other odd thing may be happening.

https://github.com/jotego/jtcores/assets/1863036/d87df313-fd0b-4de6-b286-e80efd112372

atrac17, 2023-06-11:
I agree, the scrolling in aliens isn't correct. I notice it there more than I do in scontra.

jotego, 2023-06-26:
Frame-by-frame simulation in Crime Fighters looks perfect. Maybe the odd feeling in Aliens came from object/background misalignment. This has been improved in the latest version of the core.

closed 2023-06-26 by jotego