jotego/jtcores #279: s16b: ddux Dynamite Dux color changes

jotego, 2023-06-26:
After finishing level 1, the duck color changes. Originally fixed in the old JTS16 repository (see issue [here](https://github.com/jotego/jts16/issues/10)). This was related to the DTACK signal generation.

Someone mentioned it on Patreon after the Pocket beta release of JTS16B. Not clear whether he was using an old MiSTer version.

@florgana check whether this happens on the current version in SiDi. If the problem happens, the duck color changes when he is travelling through the warp at the end of level 1. The weird colors will stay on level 2.

This was originally fixed by modifying the generation of the [DTACK signal ](https://github.com/jotego/jtframe/commit/d29e5756b220f3947a6088b8ccc797501d35494c)for the M68000. Touching the DTACK generation is dodgy because it affects games in different ways. It isn't clear why games are so sensitive to it.

florgana, 2023-06-27:
In the JTFRAME commit 1bf46f5a the duck changes colors.

An attempt has been made to increment the value to the WD parameter in jtframe_68kdtak.v and it continues to happen.

The processor speed was changed to 12 MHz and it works, the wait2 and wait3 values were also changed to 0 and it works, in both cases (independent) the duck stops changing color.

jotego, 2023-07-01:
- Moving the 16kB of main CPU RAM to BRAM does not fix it

jotego, 2023-07-06:
The DTACKN signal was generated one _cpu_cen_ too late:

Bug
[image]

Fixed
[image]

This was related to the ugly line in Ryukyu occuring later compared to MAME in #299.

Fixed in 3a724372

closed 2023-07-06 by jotego