jotego/jtcores #293: s16b: Measure the length of /VINT signal on SEGA System 16B

jotego, 2023-07-03:
Measure the length of /VINT signal, which is pin 55 of 315-5197

etriana85, 2023-07-03:
Below I attach images with the requested measurements
[image]
[image]
[image]
[image]
[image]

etriana85, 2023-07-03:
Others that may be of interest...
[image]
[image]
[image]
[image]
[image]

closed 2023-07-03 by etriana85