jotego/jtcores #305: ngp: [Baseball Stars]

jotego, 2023-07-10:
commit 6b490a07

The game halts to a white screen. No sound, no reaction to buttons.

jotego, 2023-07-12:
The code deviates from MAME at PC=FF27AE (frame 536) where MAME takes the jump and the core doesn't. 

```
PC=FF27A9,
XWA0=0,XBC0=0,XDE0=0,XHL0=0,
XWA1=0,XBC1=0,XDE1=0,XHL1=0,
XWA2=0,XBC2=0,XDE2=0,XHL2=0,
XWA3=8100CF,XBC3=20202000,XDE3=30,XHL3=6260,
XIX=FF2FCB,XIY=FF3C07,XIZ=6480,XSP=6BF0,* FF27A9: 
cp (0x6c20),0x00

PC=FF27AE,
XWA0=0,XBC0=0,XDE0=0,XHL0=0,
XWA1=0,XBC1=0,XDE1=0,XHL1=0,
XWA2=0,XBC2=0,XDE2=0,XHL2=0,
XWA3=8100CF,XBC3=20202000,XDE3=30,
XHL3=6260,XIX=FF2FCB,XIY=FF3C07,XIZ=6480,XSP=6BF0,* FF27AE: 
jr Z,0xff27e2
```
(0x6c20) is 0x2FF in the core

florgana, 2023-07-17:
The game starts about 2 minutes after showing the white screen.
The game let play but some selection texts and images are not visible. 

[image]
[image]
[image]
[image]

jotego, 2023-12-31:
The game writes to the flash before showing the title screen. Tha process seems to go fast in simulation, but not on the MiSTer

jotego, 2024-01-03:
The game itself works correctly. The flash problem is taken to a different issue

closed 2024-01-03 by jotego