jotego/jtcores #306: ngp: [NeoGeo Cup '98]

florgana, 2023-07-12:
No sound. 
The game shows 3 images blinking and then halts to white screen. 
 no reaction to buttons, no reaction to buttons.

jotego, 2023-07-15:
commit b8dff835
sound works
the game hangs up after the first three images in the intro scene

jotego, 2023-07-15:
commit 78e5acc9
the game does not hang up any more
video still too fast
colors seem inverted sometimes

jotego, 2023-12-31:
commit: 7cc0b90f
- sound ok
- speed seems ok
- general color problem
- missing bottom score row. The bottom goal is permanently shown
[image]

jotego, 2024-01-05:
fixed after adding micro DMA support

closed 2024-01-05 by jotego