jotego/jtcores #308: ngp: [King of fighters R-1]

florgana, 2023-07-12:
No sound.
Missing initial images. Game starts with flashing white screen, then written game name and stay black screen with no prompts, react with 1 (ctrl) button and show splash screen.
The select **Start,** shows another selection screen that reaction only to some number buttons, hanging the game or restarting it. After of it hangs or sometimes after reset it shows a striped screen.

[image]

jotego, 2023-07-15:
commit: b8dff835
the intro sequence run for a few seconds, with music upto this screen:

[image]

jotego, 2023-07-16:
commit 42b66755 The game works and seems stable. 

When choosing a character, the image becomes corrupted
[image]

jotego, 2023-07-16:
commit 80f26c27
fixed character selection. The game hangs up at the beginning of a battle

jotego, 2023-07-19:
commit 39fe6345
The game does hang up at the beginning f a battle. It hangs up at the end now.

florgana, 2023-07-20:
It happens like in the baseball game, after a battle it stays in white screen for a short time and then shows the option to continue the game

jotego, 2023-12-31:
Commit: 7cc0b90f
Seems playable

closed 2023-12-31 by jotego