jotego/jtcores #309: ngp: [Melon Chan's Growth Diary]

florgana, 2023-07-12:
No sound.
Doesn't display boot images correctly. The colors are wrong, it looks dark. Does not show white borders as video format in images. (img 1 & img 2)
The selection to "start" the game are not seen. (img 3)
When you start the game it lets you do some movements but after a short time it crashes and doesn't react to the buttons (img 4)

(img 1)
[image]
(img 2)
[image]
(img 3)
[image]
(img 4)
[image]

jotego, 2023-07-15:
commit: b8dff835
there is sound but the graphics look corrupted in the room now and the game hangs up soon after entering the room

[image]

jotego, 2024-09-21:
colors are ok in 132b7d0aa but it is still not playable.

closed 2024-09-21 by jotego