jotego/jtcores #318: x-men: medidas de objetos

jotego, 2023-07-31:
Objetivo:

- Entender cómo el juego y el chip de objetos usa la RAM durante los ciclos de DMA (cuando OBJ/IRQ está alto)
- ¿Varían A13~A12 según la escena o el _frame_?
- ¿A qué ritmo se lee la memoria durante DMA?

Procedimiento:

- Durante DMA (OBJ/IRQ alto) mide las señales A13~1. Si no puedes medirlas todas a la vez, prescinde de señales en mitad del bus (por ejemplo A7, A8...)
- Mide el ritmo al que A1~A3 cambia
- Deja pasar distintas partes de la demo del juego y mira si A13~A12 cambian durante DMA o si siempre tienen un valor fijo

[image]

jotego, 2023-08-01:
El orden del bus A del chip no tiene por qué coincidir con el de la CPU necesariamente. Tienen que estar ordenada de forma que durante el DMA el bus haga una cuenta ascendente. A0 tiene que ser la que conmute más rápido, luego A1, etc. 

[image]

Tiene que ser algo parecido a esto:

[image]

Prueba a reordenarlas y compara ese orden con _The Simpsons_

etriana85, 2023-08-02:
Las señales A13~12 varían si DMA esta bajo. Cuando DMA está alto, A13~12 se quedan abajo.
 
[image]
[image]
[image]

closed 2023-08-03 by etriana85