jotego/jtcores #319: The Fairy Land schematics

jotego, 2023-07-31:
Extract schematics for _The Fairy Land_ PCB

etriana85, 2023-10-06:
Work in progress

jotego, 2023-12-08:
@etriana85 acciones después de revisar la placa adaptadora:
- [ ] Revisar la placa adaptadora para entender por qué algunas huellas no tenían el tamaño correcto
- [ ] Corregir las huellas en la placa
- [ ] Reducir el largo del conector JAMMA macho para que no sobresalga tanto al encajarlo
- [ ] Quitar 1mm por cada lado al ancho del conector JAMMA macho para que encaje mejor

etriana85, 2023-12-14:
Quitar 1mm por cada lado al ancho del conector JAMMA macho para que encaje mejor: Creo que el problema no es del largo sino del ancho. 
Trabajaré hoy en la reducción del pad de JAMMA. Actualmente tiene 10mm. Te parece bien dejarlo en 7mm?

closed 2024-03-22 by jotego