jotego/jtcores #320: the simpsons: schematics for object chips

jotego, 2023-08-01:
Hoy o mañana llegará una placa de The Simpsons, préstamo del museo Arcade Vintage. Los esquemáticos de Konami carecen de la parte de los objetos. Debería ser muy similar a X-Men. Extrae el esquemático para esa sección.

etriana85, 2023-08-03:
Work in progress.

etriana85, 2023-08-08:
Adjunto esquemático de la extracción de objectos, fondo y colmix.
[The simpsons.pdf](https://github.com/jotego/jtcores/files/12292364/The.simpsons.pdf)

etriana85, 2023-08-08:
Adjunto medidas solicitadas:
[image]
[image]
[image]
[image]
[image]
[SIMPSONS.webm](https://github.com/jotego/jtcores/assets/92724765/f8169ed2-3167-42e0-8193-8f0fd0151616)
[image]
[image]

closed 2023-08-09 by etriana85