jotego/jtcores #324: karnov: blank screen

jotego, 2023-08-10:
it looks like after the recent change to jtframe_68kdtack, jtkarnov does not boot correctly (both Karnov and Chelnov). Seen in c874ac43eedf0 and probably started with the jtframe_68kdtack change.

jotego, 2023-09-03:
If [this version](https://github.com/jotego/jtframe/blob/3f54f652ae7aed0135d0607b0bdfb83eaf607558/hdl/cpu/jtframe_68kdtack.v) of jtframe_68kdtack is used, both JTCPS1 and JTKARNOV work ok

jotego, 2023-09-03:
public wokring version is 46ba76a

jotego, 2023-09-08:
- CPS1
- CPS15
- SF

Fixed in 1e5a4916. The problem was that the SDRAM was doing a write access with the write mask at 'b11 (disabled). That happened as the ram_cs was active when AS was asserted and the access was registered there but UDS/LDS are asserted later.
It looks like RW is asserted at the same time than AS, though.
The logic was changed to wait until UDS/LDS are asserted before enabling the ram_cs and other *cs signals.

asturur, 2023-09-08:
is there a way to try this commit yet?

jotego, 2023-11-03:
Karnov public version is 46ba76ad. At that time the repository used JTFRAME as a submodule, so it is hard to switch between the current commit (JTFRAME as a folder) and the old one.
Trying the old commit renders a working core on MiST, but the verilator simulation does not boot up.

jotego, 2023-11-03:
The last one with JTFRAME as a submodule still worked: 67256624

jotego, 2023-11-04:
cf58e37ccf05daa8159fc4cceda0eade76bcea1f is the commit that broken karnov

- commit cf58e37ccf05daa8159fc4cceda0eade76bcea1f (tag: karnov_bad)
- commit f6995e35f68631a04b3c16a49d269d0ad5d4302a (tag: karnov_ok)

jotego, 2023-11-04:
- just taking karnov_ok's dtack module to master's HEAD (7d159690612614e3253de879af3e73fbb13538d9) does not fix it
- simulation does not work on karnov_ok (but the FPGA does)

jotego, 2024-01-16:
Sometimes comparing with the working version is an easy way to find the problem. You can save the waveforms in the two versions and compare. If you convert a few critical waveforms to VCD, you can use `jtutil vcd compare` to find the time at which the signals split up.

These M68000 games are very sensitive to how the DTACK signal is generated. Even though the generation should be straight forward, I've found all sort of odd sensitivites over the years. I think we broke this game when we fixed Cotton in System 16B, by modifying the DTACK generation. Dynamite Dux also had a weird DTACK sensitivity #279 and so did Shinobi. So it is always scary to touch that module. Yet, the same module should work for all games. There is no technical justification for requiring two different versions of it.

gyurco, 2024-01-16:
Also there's 'we' and 'wrin' inputs, but what's the difference? I see we arrives later then wrin, then the module acts on wrin.

gyurco, 2024-01-16:
It's the write:
[image]

And the read:
[image]

The read arrives in 1 cycle, so it must be cached from the previous read, I guess. But it's not identical to the one is written.

jotego, 2024-01-17:
This core is a bit unusual in the sense that it writes to SDRAM bank 3, rather than 0. There is a macro to enable that

JTFRAME_BA3_WEN

But it might be failing. You can easily move it to bank 0 by dragging those lines in mem.yaml up to the bank 0 part, where the main CPU ROM is.

I think I tried that the last time I looked at this. But I don't remember well. Maybe it's written in this issue log. 

That's the only funny thing I can think of related to the SDRAM as writes to banks other than the zero have not been used elsewhere as far as I remember.

gyurco, 2024-01-17:
I noticed that macro. As now I went up to the top with the jtframe_ramx_xslot signals, the next step is to move into the real sdram controller.

gyurco, 2024-01-17:
BTW, CPU ROM and RAM logically should be put to the same bank, as there can be no overlapped access between the two.

jotego, 2024-01-17:
> BTW, CPU ROM and RAM logically should be put to the same bank, as there can be no overlapped access between the two.

Yes, that is the general case and the case in this core. In some cores, there are more devices accessing the same bank and it may be beneficial to have the RAM in a separate bank. I do not remember why the RAM is at bank 3 right now. Maybe I was just trying something out and it got committed for no good reason. It is better to move it back to bank 0.

By the way, remember that you can check the available macros in the command line with the command `jtmacros`. If you add part of a name, you will get only the macros containing that name `jtmacros credits`. This is convenient to consult the meaning or need of different macros.

gyurco, 2024-01-17:
So here's the thing:
[image]

Bank3's wr_busy is just one cycle before nWE on the SDRAM goes down, so this:
'''
dq_pad <= wr_cycle ? (prog_en ? prog_din : din) : 16'hzzzz;
'''
should work.

However din is double delayed, so it's not ready when it's needed. Solution: make din combinatorial in jtframe_sdram64.v. Now this jsr-rts combo seems to work, however Karnov still doesn't start properly (but the debug overlay starts to behave randomly - what's actually shown there?)

gyurco, 2024-01-17:
BTW, the Verilator and non-Verilator parts are differing as din is not delayed to sdram_din with Verilator, and delayed to sdram_dq with non-Verilator.

jotego, 2024-01-17:
That sounds familiar. But, there are many cores writing to the SDRAM. Literally a few hundred games are running that write to the SDRAM. I am a bit suspicious about that being the real culprit. The SDRAM signals are a bit adjusted for the Verilator simulation because Verilator does not do timing, it is only event driven, whereas the actual SDRAM relies uses a shifted clock. But that should not alter your results. 

I will try fixing #382 so we can re-test the controller isolatedly.

It looks like the fail occurs very early on. It may be worth running a sim to dump and evaluate all the signals. Pull the latest version and run the sim with `jtsim -setname karnov -q -video 10 -w -d VERILATOR_KEEP_SDRAM`. You will get the first 10 frames after the ROM load. If you run a ROM load once, you can skip it the next time and just do `jtsim -q -video 10 -w -d VERILATOR_KEEP_SDRAM`, which will get there much faster.

If you track the `debug_view` in jtkarnov_game you can see what each signal is doing. Usually the debug_bus will control it. Read the **debug.md** file for more details because there is other information you can also get through that debug display.

gyurco, 2024-01-17:
With the din change, the CPU looks like executing valid code, loops here and there. Maybe some IO read/write is bad now.

gyurco, 2024-01-17:
Or the din change was OK for ram writes, but not OK for other things? But nothing else writes to SDRAM...
And this makes me question what writes to obj mem? If ram_cs is for both obj and CPU RAM, then using only main_addr[13:1] would overwrite each other's area. If not, then there's nothing writing to obj RAM (spriteram in MAME).

gyurco, 2024-01-17:
Looks like objram_cs for writes goes to BRAM? Then why obj_cs goes to jtframe_ram1_2slots?

gyurco, 2024-01-17:
BTW, the game has music, and it seems to run...just without any graphics.

gyurco, 2024-01-17:
I just died three times, I got endgame tune :D Now it's in demo mode...with a full black screen.

gyurco, 2024-01-17:
Ah, obj_cs is for tilemap data...then probably it's not written into bank3 during programming.

gyurco, 2024-01-17:
Here's the diff which makes the CPU RAM OK:

```
diff --git a/modules/jtframe/hdl/sdram/jtframe_sdram64.v b/modules/jtframe/hdl/sdram/jtframe_sdram64.v
index 325bffe8..c5ba7efc 100644
--- a/modules/jtframe/hdl/sdram/jtframe_sdram64.v
+++ b/modules/jtframe/hdl/sdram/jtframe_sdram64.v
@@ -136,7 +136,7 @@ reg  [14:0] prio_lfsr;
 wire [12:0] bx0_a, bx1_a, bx2_a, bx3_a, init_a, next_a, rfsh_a,
             ba0_row, ba1_row, ba2_row, ba3_row;
 wire [ 1:0] next_ba, prio;
-reg  [15:0] din;
+wire [15:0] din;
 
 wire [AW-1:0] ba0_addr_l, ba1_addr_l, ba2_addr_l, ba3_addr_l;
 wire    [3:0] rd_l, wr_aux;
@@ -198,6 +198,10 @@ always @(negedge clk) begin
     other_rst <= prog_en | init | rst;
 end
 
+assign din = (bg[3] && BA3_WEN) ? ba3_din :
+             (bg[2] && BA2_WEN) ? ba2_din :
+             (bg[1] && BA1_WEN) ? ba1_din : ba0_din;
+
 always @(posedge clk) begin
     dst      <= ba_dst;
     rdy      <= ba_rdy;
@@ -217,9 +221,6 @@ always @(posedge clk) begin
     sdram_a[10:0] <= next_a[10:0];
 
     wr_l <= wr_aux;
-    din <= (bg[3] && BA3_WEN) ? ba3_din :
-           (bg[2] && BA2_WEN) ? ba2_din :
-           (bg[1] && BA1_WEN) ? ba1_din : ba0_din;
 
 `ifndef VERILATOR
     dq_pad <= wr_cycle ? (prog_en ? prog_din : din) : 16'hzzzz;
```

Meanwhile obj_cs is never asserted during the running game, obj_addr is always 0. Maybe VRAM/Sprite RAM writes are bad (but they're in BRAM).

jotego, 2024-01-17:
Very good progress! If look at the log:

> https://github.com/jotego/jtcores/commit/cf58e37ccf05daa8159fc4cceda0eade76bcea1f is the commit that broken karnov
>
>    commit https://github.com/jotego/jtcores/commit/cf58e37ccf05daa8159fc4cceda0eade76bcea1f (tag: karnov_bad)
>    commit https://github.com/jotego/jtcores/commit/f6995e35f68631a04b3c16a49d269d0ad5d4302a (tag: karnov_ok)

And if you compare those two commits, the sdram controller was not touched. Indeed the only common file was the DTACK generation. Maybe the new DTACK unveiled a weakness in the controller?

Did you check that Verilator sims still work after the change? Do you get video in the sims?

gyurco, 2024-01-17:
I got this with f6995e3:
panic: open /home/gyuri/git/jtcores/modules/jtframe/hdl/inc/game_sdram.v: no such file or directory

I didn't check anything with simulator yet.

jotego, 2024-01-17:
Recompile jtframe each time you change to an old (or new) commit. Just type jtframe and it should recompile

gyurco, 2024-01-17:
Well, if dsn is low active, then this:
```
assign objram_we  = {2{objram_cs}} & main_dsn;
```
won't work.

gyurco, 2024-01-17:
This one does the job well:

```
diff --git a/cores/karnov/hdl/jtkarnov_game.v b/cores/karnov/hdl/jtkarnov_game.v
index ef8b4f73..af60e211 100644
--- a/cores/karnov/hdl/jtkarnov_game.v
+++ b/cores/karnov/hdl/jtkarnov_game.v
@@ -38,9 +38,9 @@ assign debug_view = debug_bus[7] ? st_main : mcu_st;
 assign ram_we     = ram_cs & ~main_wrn;
 assign dma_we     = {2{pre_dma_we}};
 
-assign vram_we    = {2{vram_cs  }} & main_dsn;
-assign scrram_we  = {2{scrram_cs}} & main_dsn;
-assign objram_we  = {2{objram_cs}} & main_dsn;
+assign vram_we    = {2{vram_cs   & ~main_wrn}} & ~main_dsn;
+assign scrram_we  = {2{scrram_cs & ~main_wrn}} & ~main_dsn;
+assign objram_we  = {2{objram_cs & ~main_wrn}} & ~main_dsn;
 
 // Remove this when bus contention is done
 assign sdtkn = 0;
```

Now Karnov kills!

gyurco, 2024-01-17:
Looks like the sim also produces something:
[image]

gyurco, 2024-01-17:
What I noticed in the game that sprites are 'dancing' during horizontal scroll. I saw similar thing in Sega System1, where the sprite X position had to be rounded:
https://github.com/Gehstock/Mist_FPGA/commit/983ede0e345d358c83b712f6de4c91259cb9abf2#diff-537cb8e07f29e5147253d24fb2ed61daf1e312a4194312082bf05692ba4f4309L150
(line 150)

jotego, 2024-01-18:
Excellent news!

The *_we assignments were wrong indeed. Looking at the git log, I think the write signals had been merged in the _mem_ template before and when I consolidated all repos into one, this mismatch between the _mem_ template and the core occured and this regression error appeared. That's why I've been trying to have regression tests for a year! Thanks a lot for finding this. DTACK had nothing to do in the end.

The sprite dancing should be related to when the object table buffer is updated. I think that rounded the X position is the wrong solution. Games normally write the sprite data to a table that needs to be copied into the rendering stage at a specific time. What controls that is system dependent. Also, some games in some systems do it wrong due to bad software, not bad hardware. Once we close this issue, I think it's worth it opening a different one to review the sprite DMA.

jotego, 2024-01-18:
@gyurco, could you check if #409 still occurs? I think it must have been related to the missing signals in the write.

gyurco, 2024-01-18:
> @gyurco, could you check if #409 still occurs? I think it must have been related to the missing signals in the write.

It looks okay now.

closed 2024-01-18 by jotego

jotego, 2024-01-19:
Simulation is not working for me for karnov. Can you double check on your end please?

gyurco, 2024-01-19:
It still works with current master:
```
Verilator sim starts
Multibank SDRAM enabled
WARNING: (test.cpp)sdram_bank0.bin not found
WARNING: (test.cpp)sdram_bank1.bin not found
WARNING: (test.cpp)sdram_bank2.bin not found
WARNING: (test.cpp)sdram_bank3.bin not found
Read 1284096 bytes from rom.bin
Simulation clock period set to 20832 ps (48.003072 MHz)
Verilator will dump to test.vcd
DIP sw set to FFBF
WARNING: TOP.game_test.u_game.u_game.u_mcu.u_prom cannot open file: ../../../../rom/chelnov/ee-e.k14
WARNING: TOP.game_test.u_game.u_bram_vram.u_lo.u_ram cannot open file: vram_lo.bin
WARNING: TOP.game_test.u_game.u_bram_vram.u_hi.u_ram cannot open file: vram_hi.bin
WARNING: TOP.game_test.u_game.u_bram_scrram.u_lo.u_ram cannot open file: scrram_lo.bin
WARNING: TOP.game_test.u_game.u_bram_scrram.u_hi.u_ram cannot open file: scrram_hi.bin
WARNING: TOP.game_test.u_game.u_bram_objram.u_lo.u_ram cannot open file: objram_lo.bin
WARNING: TOP.game_test.u_game.u_bram_objram.u_hi.u_ram cannot open file: objram_hi.bin
jtframe_pxlcen: using 4 as clock divider
SDRAM burst mode changed to 2 mask 0xFFFFFFFE -> burst per bank = { 2, 2, 2, 2 }
```

jotego, 2024-01-19:
Did you check the output in the frames folder? I only get a blank first frame.

gyurco, 2024-01-19:
I got two files, frame_00001 and frame_00027, the first is blank, the second is good (high scores on it).
I don't know why just two files...the one I posted above was frame_00009 I think. Looks the sim chooses two random frames?

jotego, 2024-01-19:
By default, you only get a new image file when it is different from the previous one in order not to flood the folder and your eyes with irrelevant information.

I am stuck at frame_00001. Can you please try going through the load process: `jtsim -setname chelnov -video 100`?

gyurco, 2024-01-19:
BTW, I copied the .rom file by hand to jtcores/rom, as it wrote 'cannot produce chelnov.rom'
Running with bash -x, it just revealed that it just looking for the .rom, but it doesn't want to 'produce' it.

jotego, 2024-01-19:
> BTW, I copied the .rom file by hand to jtcores/rom, as it wrote 'cannot produce chelnov.rom' Running with bash -x, it just revealed that it just looking for the .rom, but it doesn't want to 'produce' it.

Things to check:

- always source setprj.sh in your current terminal (or source it in .bashrc)
- `jtframe mra` looks for MAME zip files in `~/.mame/roms`

`jtframe mra corename` will create the .rom file. No need for the `mra` tool. When calling `jtsim -setname xxx` it will run `jtframe mra` for you, generate the file in $ROM and link it.

gyurco, 2024-01-19:
The issue was I had ~/mame/roms, now I have ~/.mame/roms, too. ROM files are generated.
I get this for frame_00036:
[image]

jotego, 2024-01-19:
Thank you. I'll check it further on my end.

gyurco, 2024-01-19:
Just a question: what macro adds core_mod[2] to the MRA? I think it's missing in 1943, even the OSD is rotated in the wrong direction with current master (and old MRA). Comparing with 1942 (which has 5) I don't see any obvious macro to affect this.

jotego, 2024-01-19:
## core_mod

Bit    |  Use                      | Set by
-------|---------------------------|---------
0      | High for vertical games   | mame.xml
1      | 4-way joysticks           | JTFRAME_SUPPORT_4WAY
2      | XOR with dip_flip         | mame.xml
3      | dial input enable         | mame.xml
4      | reverse the dial          | mame.xml

Most core_mod bits are derived from MAME. Some games produce a flip bit that means the opposite as in MAME. In that case `dip_flip` must be assigned to `~flip`. Making sure everything is consistent is a bit of a pain that pops up from time to time. When things are right, you get the right orientation in MiST's OSD and the right orientation in MiSTer's rotation and simulation. All should agree if the core produces the right dip_flip value.

If there's something to fix, let's move it to a different issue.