jotego/jtcores #327: The Simpsons 2-player - ROM check fail after saving soft dips

sqwirral, 2023-08-11:
MiSTer.

Enter service menu with F2, make changes, then do Save Settings in OSD. 
After reloading core, booting fails because ROM 12c gets flagged as bad. Need to delete .nvm file to boot again.

Happens in all the 2-player MRAs I tested (Japan, World 1, 2, 3). Did not happen in 4-player (but the soft dips still did not save).

sqwirral, 2023-08-23:
This also happens when adjusting the CRT offset in the OSD.

I set horiz and vert to +4, Save Settings, and reload core. Same 12C ROM chip error on boot. No NVRAM file was even created. Deleting the .cfg file fixes it.

jotego, 2023-08-25:
- NVRAM file dumps are created with an address offset of +1 on MiSTer. MiST is ok.
- NVRAM files read process works correctly

There was a problem in hps_io for 8-bit access. Fixed in d5cca054

jotego, 2023-11-09:
This seems to be happening again in the Halloween beta update. 

- MiSTer version works
- Pocket save files appear corrupted

The Pocket save file content does not match MiSTer's one

- [ ] Check that the Pocket can load the MiSTer file - isolate loading and saving procedures
- [ ] try saving ioctl_addr as data, to isolate the ioctl_din generation from the JTFRAME side of the saving

jotego, 2023-11-14:
Setting ioctl_addr as ioctl_din in jtsimson_game `assign ioctl_din  = ioctl_addr[7:0];`, shows that ioctl_din is ignored:

```
00000000: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000010: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000020: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000030: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000040: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000050: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000060: 0300 0000 0300 0000 0300 0000 0300 0000  ................
00000070: 0300 0000 0300 0000 0300 0000 0300 0000  ................
```

jotego, 2023-11-14:
Fixed in 25026ec1

closed 2023-11-14 by jotego