jotego/jtcores #329: simson: scene 17

jotego, 2023-08-12:
Fixed in 8568b5c9

asturur, 2023-08-20:
Hi!
Not sure what scene17 is, but yesterday i played all the game and after the boss battle on the japanese stage ( the tv studio one ) i got some glitches on screen while moving forward to the 2 final boss battles.

I never see how the real hardware behave there since i never got that far.
I was playing bart in the 2 player version of the game.

jotego, 2023-08-21:
This is fixed but I have not released the new file yet.

asturur, 2023-08-21:
So scene 17 is this particular one?

jotego, 2023-09-01:
This is scene 17

[image]

It looks like it still happens a bit during gameplay, allegedly improved. The scene simulation is fine but that has less SDRAM overhead than the actual core.

jotego, 2023-09-02:
I have been able to speed up the sprite logic a bit, I can see that it has more margin in simulation. 

I have updated the files in JTBIN. Please let me know if it still occurs.

closed 2023-09-02 by jotego