jotego/jtcores #341: The Final Round schematics

jotego, 2023-08-27:
- [ ] Change A[0..12] to A[1..12]

[image]

- [ ] Add number to A bus in symbol. Keep the same order as in the D bus

[image]

- [ ] Align U176 and U177 vertically to make the signal path more obvious for all 8 bits of U5

[image]

etriana85, 2023-08-28:
Done!
[image]
[image]
[image]
[image]
[image]

etriana85, 2023-08-28:
[image]

jotego, 2023-08-29:
Another one:

- [ ] change to De Morgan (sprites sheet)

[image]

etriana85, 2023-08-29:
Done!
[image]
[image]

closed 2023-08-29 by jotego