jotego/jtcores #348: Splatter House: measure video signals

jotego, 2023-09-01:
Medir en Splatter House las señales VSYNC, HSYNC, VBLANK, HBLANK para que esté clara su duración, polaridad y posición relativa.

Medir la frecuencia del reloj, que debería ser 49.152MHz pero puede haberse corrido con el tiempo

[image]

Son los mismos esquemáticos que PacMania

etriana85, 2023-09-08:
[image]
[image]
[image]

etriana85, 2023-09-08:
~VBLANK Lo veo siempre bajo (CERO) y ~VSYNC alto  (UNO).
EL RELOJ: 46.5 MHz ~ 50 MHz. Pero esta en el pin 33.

jotego, 2023-09-19:
No hay bastante información en las gráficas para reproducir las señales HB/HS. Es imposible saber el periodo de las señales con esas gráficas o la longitud de la parte baja de HS y HB

Es necesario medir el reloj con una precisión de al menos 1%

VB/VS tienen que funcionar o no habría imagen en el monitor

etriana85, 2023-09-25:
[image]
[image]
[image]
[image]

etriana85, 2023-09-25:
El reloj: 49,1 MHz 48.8 MHz

He corregido las señales, las he tomado de un chip personalizado (CUS133 - pág. 11).  Y en cuanto el reloj, lo había seguido erroneamente y estaba midiendo el pin equivocado.

Quedo atenta a cualquier novedad.

etriana85, 2023-09-25:
[image]

etriana85, 2023-09-25:
- pulsos para medir H de CLK12,2MHz
- HS activo 64 pulsos
- HB activo 192 pulsos
- Periodo HS/HB 768 pulsos
- Retraso de activación HB a activación HS 64 pulsos

- pulsos para medir V de HS
- VS activo 8 pulsos
- VB activo 40 pulsos
- Periodo VS/VB  264 pulsos
- Retraso de activación VB a activación VS 8 pulsos

jotego, 2023-09-25:
Todo correcto. Gracias

closed 2023-09-25 by jotego