jotego/jtcores #349: Ghosts'n Goblins errors leading to crash

Aquas77, 2023-09-09:
Footage of potentially same issue:
First footage from RoboyZHunter: https://www.youtube.com/watch?v=qwmFzKJxYbI&feature=youtu.be - (playing World version I believe.)
Footage from me: https://www.twitch.tv/videos/1920550515 - (J, Rev G version)

Report from RoboyZhunter: "the first thing you will see is that the timer stops working if that timer stops working then there is a malfunction. If I die or complete a level while the timer is doing this buggy stuff then  the game will basically just crash and I have to press reset and start a run all over"

About my experience and his:
"Both of us has the in game timer pausing then when we enter a level the game would either let us play one more round or it will crash once we either die or enter a new stage
For Aquas they have the take the key text on screen for a bit and then once they beat the level the game totally crash
Both of us seem to be on the same page with this glitch in that it typically starts when the timer stops working and then the game decides on when it will crash from there"

Furthermore, in my footage, the timer stops at first, then the score begins to tabulate only periodically, then the Take a Key for Coming In text remains on screen for the next stage, remains playable, then crashes on stage complete.

I have updated my MiSTer recently but this was my first experience with this problem. But RoboyZ has reported that this problem has occurred for him since April of 2022. 32MB ram on my end, 128 on Roboy's.

jotego, 2023-09-09:
I left the core running overnight and indeed it had crashed by the morning (commit 46ba76a). Let me review it.

jotego, 2023-09-12:
Confirmed on most recet commit too 0257e01a

jotego, 2023-09-16:
_Modernizing_ the code with the current practices in JTFRAME around the CPU did not fix the problem in 3439fefb. Still, after many hours, the game attract mode will hang up

jotego, 2023-09-17:
The game consistently hangs up after many hours in attract mode at the same point:

[image]

Using signal tap, the CPU seems to move around the loop in $617F and triggers the object DMA normally. S and CC register look normal and match MAME's values.

The attract loop is not fixed but seems to be pseudo random.

Toryalai1, 2023-09-19:
This is apparently a real bug in hardware and not a core bug. Here is a post about it from an X user:
https://x.com/TheRealPatete/status/1142826078987268098?s=20

ksdomino, 2023-09-19:
Well there are 7 revisions. Might be worth testing in mame to see if the the best/latest ones (US) and world 3 ( according to : https://tcrf.net/Ghosts%27n_Goblins_(Arcade)#Revisional_Differences ) have the same behaviour/crash.

jotego, 2023-09-19:
The world one crashes in MAME too, as I said above. The Japanese version G works correctly after more than 24 hours on MiSTer.

sqwirral, 2023-10-01:
Another recorded occurance of the crash here. You see the timer stop beforehand.
https://www.twitch.tv/videos/1939440306?t=2h30m04s

jotego, 2023-10-02:
@Aquas77 when you try the new version -which should be online tomorrow- please play with the yellow flash on. Turning it off block CPU writes to the palette RAM during the visible part of the frame. This could potentially cause a crash if the CPU tries to read that data back.

jotego, 2023-10-04:
Released in JTBIN ([commit link](https://github.com/jotego/jtbin/tree/31bc8232de75beb61d6c98beaad0b19aec8e8763)). Please let me know how it goes.

closed 2023-10-04 by jotego

RoboyZHunter, 2023-10-28:
Does it matter which revision you play to prevent this crash?