jotego/jtcores #375: gng: make mister/pocket release

jotego, 2023-10-02:
Recompilar y lanzar gng sólo para Pocket y MiSTer

```
cd $JTBIN
git pull
cdjt
jtupdate gng -t mister,pocket --git
cd $JTBIN
jtbin2sd --git
jtbin2mr --git
```

Comprobar:

- [ ] Pocket
- [ ] MiSTer
- [ ] Los ficheros modificados en JTBIN tendrían que ser sólo los de GNG, y no otros cores
- [ ] Los DIP switches tienen que funcionar bien. Comprobar en el modo de servicio que la configuración que se ve en el OSD es la que ve la máquina

Si todo está correcto:

```
cd $JTBIN
mergenkill
git push
```
Y marcar #349 como "Done" en el apartado de _Projects_

florgana, 2023-10-02:
Done! All okay.

jotego, 2023-10-02:
There's no commit in [JTBIN](https://github.com/jotego/jtbin/commits/master). Did you forget to _git push_ it?

jotego, 2023-10-04:
released in https://github.com/jotego/jtbin/tree/31bc8232de75beb61d6c98beaad0b19aec8e8763

closed 2023-10-04 by jotego