jotego/jtcores #376: gng: video not supported on MiST

jotego, 2023-10-02:
The video output on MiST has stopped working. The monitor shows a blue screen. @florgana, please try it on SiDi version: is there image?

florgana, 2023-10-02:
Commit 12f3282a 
In SiDi the monitor also shows a blue screen, but if it generates sounds when executing some action. This means that the core works but does not generate video.

jotego, 2023-10-02:
The scan doubler may be disabled. 

- check if there is some status bit in MiST that would disable the scan doubler.

jotego, 2023-10-02:
@florgana try compiling with this line:

https://github.com/jotego/jtcores/blob/28d7a8e97002d95a765a2963822defb0c2d80316/modules/jtframe/target/mist/user_io.v#L124

set `assign scandoubler_disable = 0;`

Do not commit that change. It is just to understand what can cause this problem.

jotego, 2023-10-03:
- Connect the oscilloscope to the VGA output pins in SiDi. 
- Compare the output of the current broken version, and the working one in JTBIN/sidi
- Measure pins H-SYNC, V-SYNC and fill this table

Pin | Frequency old | Frequency new |
------|--------|----
commit | | 
H-Sync |   .. kHz  | .. kHz
V-Sync  | .. Hz | .. Hz
RGB | Yes | activity? Yes/no |

[image]

florgana, 2023-10-04:
We connected the oscilloscope to the output pins on SiDi and the result is very similar

Pin | Frequency old | Frequency new |
------|--------|----
commit | jtbin | bfca0392 
H-Sync |   31.3 kHz  | 31.2 kHz
V-Sync  | 59.7 Hz | 59.7 Hz
RGB | Yes | Yes |

I looked for the commit that could have caused the error and found that in commit 62b657cf0711e75b3acbe9e8c3ce8a9550740271 it works. and the next commit f43b2b5f11030ee374f1401e866f388cdc4546f9 does not generate video.

jotego, 2023-11-01:
The image is visible through OSSC but it is highly unstable and wavy (latest commit tried 255d879e)

jotego, 2023-11-01:
jtgng_timer was fed the cen from a different clock divider than the video, as one was coming from jtframe_board and the other from the game module

closed 2023-11-01 by jotego