jotego/jtcores #398: bubl: make full implementation of 6801

jotego, 2023-10-20:


LordBarker80, 2023-10-30:
Yes. Original board has less slowdown than the core. Have played several times to confirm. It's immediately noticeable from stage 1 but more prominent on stage 3 onwards.

LordBarker80, 2023-10-30:
Video link for Jotego, overview of my board as requested. Can slow down or get specific chip images: https://youtu.be/Lzrhr7iXGaw?si=3FcO-8rB36KwJLMk

LordBarker80, 2023-10-30:
As requested video overview of my originak untouched Japanese board. Let me know if you need any stills/more videos or better quality. I shot I'm 4K but unsure if it let me upload it that. https://youtu.be/FzFQJnvPC_4?si=LqKP4_KzXCjfAOxs

jotego, 2023-10-31:
On a surprising plot twist, a closer inspection showed that the MCU might have been running too fast in the core. This by itself could crease a slowdown if some timeout is triggered in the MCU firmware because of it. I'll release a new version today. Let me know how that one feels.

Toryalai1, 2023-11-01:
@LordBarker80 have you tried latest update?

LordBarker80, 2023-11-02:
Haven't had a chance to try this yet due to work. Didn't realise Jotego has released the update already! Will try tomorrow and report back.

LordBarker80, 2023-11-02:
@jotego so I updated this tonight and played through the first 10 levels. I test the PCB on one cab and then move over to the other cab with the MiSTer in so I can do either side by side or one after the other comparisons as I've done before. 

I need to give it a longer playthrough, but initially it appears the slowdown has decreased. I will test this however and aim to record videos. 

The one issue that remains however is a random paused frame. What I mean by this is there's a random pause in the action for a split second (perhaps a single frame?) that happens completely randomly. For example on my first MiSter play through the pause happened at stage 8. On my second playthrough I went a couple of stages beyond 10 to stage 12 and no pause occurred. I've tested the PCB many, many times previous to tonight and cannot get the arcade board to exhibit this behaviour. I'm going to do a more thorough test this week, but I think the slowdown is almost gone, just not quite 100%, but I need to record videos and hopefully can point out this random frame pause too.

jotego, 2023-11-03:
That pause happens on some modern screens due to the way they adjust the input signal frequency. It is only relevant to the core if you see it using a 15kHz native CRT screen.

LordBarker80, 2023-11-03:
> That pause happens on some modern screens due to the way they adjust the input signal frequency. It is only relevant to the core if you see it using a 15kHz native CRT screen.

As I mentioned above ALL of my testing is done on a CRT. I use MS9 CRT monitors in my arcade cabs and a Sony PVM for my MiSTer console setup. 

As such this is happening on a 15khz native CRT.

dazzer69, 2023-11-04:
Hi JT. 
I can confirm I’m also seeing the random pause on CRT. For what it’s worth it is something that has crept in in the last few updates. 

Keep up the good work.

LordBarker80, 2023-11-04:
@jotego So I've recorded two video of MiSTercade core Vs. PCB and I've actually found nother issue which wasn't directly present in earlier versions of the core in the same way. Let me know if you have issues accessing any of these video links. I talk quite quickly and currently have a cold, so apologies if anything isn't clear.

Video 1 - I synced up the intro/attract mode of Bubble Bobble from PCB and MiSTer running next to each other in two identical cabs and the MiSTer gradually runs behind the PCB and falls out of sync, in different way to how it did whne I first reported. That video is here: https://youtu.be/QHY75vDncEg

Video 2- This has me playing the PCB and then the MiSTer core back to back on two identical cab setups. There is still slowdown present in the MiSTer core that doesn't exist on the PCB. It also - at points - feels much more laggy on MiSTer via MiSTercade. Other cores, specifically the CPS1 and 2 cores feel exactly like the boards, I've tested side by side for them too. It seems to be the increased slowdown when the screen gets busy on the MiSTer core or sometimes when you jump, oddly, that causes the issue. That video is here: https://youtu.be/QViBos3D1g0

jotego, 2023-11-04:
It looks like there is no way around. I will have to make my own MCU module and discard the old one I found in the internet.

jotego, 2023-12-04:
@LordBarker80 I have it fixed! :-) I have used your video recording and the new version stays in sync through the whole video. I had to write my own MCU module and designed it so it is cycle accurate by construction. I will tell the story in this Friday's update!
Thanks a lot for your detailed reports. Your help has been vital to get this job done.

LordBarker80, 2023-12-04:
@jotego amazing! Will the new core download automatically today with update all? Or do we need to wait until Friday?

LordBarker80, 2023-12-04:
@jotego hey man, so have spent 45 minutes with the new core and sadly it still isn't correct :(. It now has a different behaviour. The monster patterns on the first level do not match the PCB, whereas before they did! I have tried every rom version in the alts folder and they all display the same behaviour. I have recorded 2 x cold boots of the 2 x arcade machines, the left is PCB, the right is my MiSTercade and each time they display the behavious: different monster patterns on the first levels was actually wasn't present in your original core. I'm uploading the video now. I'll also attach an OSD image to confirm it is the new core (red OSD).
[image]

jotego, 2023-12-04:
Good test! Thanks for the quick feedback. The new MCU likely has some buggy instruction -cycle accurate though :-P
How about the demo loop? Is it the same?

jotego, 2023-12-04:
I'm aware of the sound. It's a regression error introduced by a change on a different module. That's understood.
It looks promising so far!

jotego, 2023-12-05:
Yes, please. If you can start a game so I can synchronize with the coin/start screen, that'd be perfect. Then just leave it alone as you did in the other video.

jotego, 2023-12-06:
Thank you! Can you take a picture of your DIP switches? I think the settings are different from the default configuration in the core.

LordBarker80, 2023-12-06:
> Thank you! Can you take a picture of your DIP switches? I think the settings are different from the default configuration in the core.

@jotego hopefully this is clear enough. I've seen the U.S dip switches mapped online but they don't match these ones. Unless I'm mistaken. 
[image]
[image]
[image]

LordBarker80, 2023-12-06:
Just checked against the game and a set I found online and seems the same to me. All dips off is default.

jotego, 2023-12-06:
Thank you. There were a good bunch of bugs in the new MCU module. Fortunately, the whole point of the new design was to make it easy to develop and debug so it has been rather quick to lint the code. The monster pattern is good and the video speed is matched. Give a go to the attached file :-)
[release.zip](https://github.com/jotego/jtcores/files/13588328/release.zip)

Another issue is that I cannot match the first bonus item that shows on the screen between your video and any game version (either on the FPGA or MAME).

[image]

Depending on the DIP settings for difficulty and the game ROM, I can see different items popping up, but not yours. I wonder if your ROM version is different from the ones known. Maybe it is just that the GFX ROMs are different and the items display differently?

[image]
[image]
[image]

Another possibility is that the MCU is using some of its peripherals, which neither MAME nor I have implemented. See [this note](https://github.com/mamedev/mame/blob/d2e8df2916e8b0d288be6d64a609b793551cdba3/src/mame/taito/bublbobl.cpp#L260). I will be implementing them tomorrow, just in case.

LordBarker80, 2023-12-06:
> Thank you. There were a good bunch of bugs in the new MCU module. Fortunately, the whole point of the new design was to make it easy to develop and debug so it has been rather quick to lint the code. The monster pattern is good and the video speed is matched. Give a go to the attached file :-) [release.zip](https://github.com/jotego/jtcores/files/13588328/release.zip)
> 
> Another issue is that I cannot match the first bonus item that shows on the screen between your video and any game version (either on the FPGA or MAME).
> 
> ![imagen](https://private-user-images.githubusercontent.com/1863036/288438027-8dc6e6b4-52f2-4e54-8426-257552bf93a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NzkxNjQsIm5iZiI6MTcwMTg3ODg2NCwicGF0aCI6Ii8xODYzMDM2LzI4ODQzODAyNy04ZGM2ZTZiNC01MmYyLTRlNTQtODQyNi0yNTc1NTJiZjkzYTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTYwNzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWFjMTBjMmE0NmUzN2NkNTA1MTdhODNiYzllNWRkZmY5ZDAxYmFkNzUxYWMwMzBjNDYzYmI1M2IxMDgwZWVlYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.F-mdNoG4VDZrO_KLZ5iXKCDAWIuFOEs0Rx72JHWAwo4)
> 
> Depending on the DIP settings for difficulty and the game ROM, I can see different items popping up, but not yours. I wonder if your ROM version is different from the ones known. Maybe it is just that the GFX ROMs are different and the items display differently?
> 
> ![0002](https://private-user-images.githubusercontent.com/1863036/288438540-dfbdc232-0de6-46bd-8cac-2fdafd675bc2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NzkxNjQsIm5iZiI6MTcwMTg3ODg2NCwicGF0aCI6Ii8xODYzMDM2LzI4ODQzODU0MC1kZmJkYzIzMi0wZGU2LTQ2YmQtOGNhYy0yZmRhZmQ2NzViYzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTYwNzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjY5MGFiZDcxNWZiZDdhYzVlMzE4MjBhNTI0NTc0YzM3YTgxYTcyYjljMmNmMDNmODgyOTFmNTc2NTdiZTE1YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yhB-wLAvF6UdDTZBJh9kLkogdppWrsCP5gwNf8HYO5g) ![0003](https://private-user-images.githubusercontent.com/1863036/288438547-cfde0d29-3e0a-4608-aa4c-f673469103e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NzkxNjQsIm5iZiI6MTcwMTg3ODg2NCwicGF0aCI6Ii8xODYzMDM2LzI4ODQzODU0Ny1jZmRlMGQyOS0zZTBhLTQ2MDgtYWE0Yy1mNjczNDY5MTAzZTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTYwNzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzcyNTY0YjQ1NmFhOGRhODI1MDE1Y2Y3ZTM0MzZlNTJlYzg0MjA1ZGZiZjc2NWI1MzRlMmNiMjgwOTA3NGRmMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.r4DPvGBWN0ZbzxeBTEc8P955stdBObKqXABnrziSph0) ![0000](https://private-user-images.githubusercontent.com/1863036/288438594-531405fe-6303-4d59-b152-7bdc7c0ff3d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE4NzkxNjQsIm5iZiI6MTcwMTg3ODg2NCwicGF0aCI6Ii8xODYzMDM2LzI4ODQzODU5NC01MzE0MDVmZS02MzAzLTRkNTktYjE1Mi03YmRjN2MwZmYzZDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDZUMTYwNzQ0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTBjYjU4ZDNlMWYzYjVlNWMzNTVmZTY3NWU5YmNkZjcyOGMwZjRjODdlZGQzMzkzMTNlNTJlNzQ3MjdkZjdiYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Ya1mN2dD5tEY-YmiVjMA6PMoVdJ62my73yNOujZJpAs)
> 
> Another possibility is that the MCU is using some of its peripherals, which neither MAME nor I have implemented. See [this note](https://github.com/mamedev/mame/blob/d2e8df2916e8b0d288be6d64a609b793551cdba3/src/mame/taito/bublbobl.cpp#L260). I will be implementing them tomorrow, just in case.

Great. I will try this shortly! I mention in my video the bonus items will always be random, at the end of the long one I think :) this is based on the MCU and how you can tell that part is working correctly! The game is very clever it's bonus items appear based on the previous game, so they will always be different betwee  different boards and gamws. It somehow 'stores' the data of each playthrough! On very early versions of MAME the item was always the same, as this was always a problem with earlier versions of MAME emulation! So you've got it correct as far as the bonus items go :). Will check this new version shortly.

jotego, 2023-12-06:
Mmm... all the devices in the core are allegedly cycle accurate now. The board may be running at slightly different speed because of ageing in the crystal oscillator. Indeed, when I measure video timings in my boards I bypass the board oscillator to avoid that effect. If the crystal frequency has shifted by 0.8% in these 40 years, that would mean a 5 second difference in a 10 minute video.

If everything in the board was derived from one crystal oscillator, the slowdowns should match, even if the video eventually develops a visible lag. But, there are two crystals in this system, one for the three Z80 CPUs and the video, and another one for the MCU. They might have drifted in even different directions (one becoming faster and the other slower). So a small difference in the slowdowns could be explained by this too.

The next step would be to replace the two oscillators (quartz crystals and elements around them) in the board. I believe that is the next most logical culprit.

LordBarker80, 2023-12-06:
> Mmm... all the devices in the core are allegedly cycle accurate now. The board may be running at slightly different speed because of ageing in the crystal oscillator. Indeed, when I measure video [...]

Thanks Jotego. Surely though if my crystal speed has changed that wouldn't account for the slowdown in game in your core? Just to be clear there is visible slowdown when the screen gets busy in your core which does not happen on my PCB. I can understand how the crystal might affect the attract loop but can't see it getting rid of slowdown in game?

LordBarker80, 2023-12-06:
Good to know. Thanks for the info.

jotego, 2023-12-06:
[TI](https://www.ti.com/lit/an/snla290/snla290.pdf?ts=1701874459092&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FDP83825I) gives a typical figure of 5ppm/year, which would be 0.02% in 40 years. Not a lot. We'll measure some of the boards to see how much we are really getting in these old boards.

LordBarker80, 2023-12-06:
@jotego another video with the latest build. It seems it's the character behaviours that knocks it out of sync with the board. In the video you'll see one demo scene on the MiSTer core runs longer with different character patterns so as such then runs 'behind' the board. Having had a friend run his Japanese PCB, his seems to sync with mine. What would affect the character patterns I wonder?

I had a 45 minute playthrough of the new core it does 'feel' more like the board and there are no screen pauses anymore, which is great. It's mainly the  increased slowdown and ofc the intro loop which is different. 

Video: https://youtu.be/1ZhZ3twyb_k?si=Cl1hbs7HerFdhX8E

Thanks as always buddy.

jotego, 2023-12-08:
Thanks for reviewing it again! :-)
I have found another couple of bugs in the new MCU. I think this will be the last findings. I will publish the core later. Let me also tell the story of our interaction as I think people will like to know about it.
We measured a bunch of boards in the lab (see #448) but found that crystal ageing was too low to make a difference. So that is discarded as a culprit.
The monster patterns are the critical thing indeed. As an example, the NEG instruction, which changes the sign of a number, was wrong. Instead of converting -3 to +3, it would do -3 to +4, for instance. I have fixed that. Hopefully that will be it.
The previous pattern error had to do with an arithmetic operation too.

LordBarker80, 2023-12-08:
> Thanks for reviewing it again! :-) I have found another couple of bugs in the new MCU. I think this will be the last findings. I will publish the core later. Let me also tell the story of our [...]

Amazing. Thanks for your work on this. Yes, please do tell people of our intereractions, no problem. Will the newest test be released later with update all or is there a new one I can test this afternoon? I have soon time this afternoon so I can test if needed.

LordBarker80, 2023-12-09:
@jotego hey man, the intro sync seems to be out, the exact same as the test build on December 6th. This video I previously linked: https://youtu.be/1ZhZ3twyb_k?feature=shared    goes out of sync at 5 minutes due to the monster and character patterns. This is still the same in the latest build. The slowdown, sadly, is still present in levels. When the screen gets busy the slowdown on the MiSTer core increases more than the arcade. There is minimal to no slowdown on the board compared to the core. An easy way to see this is - how I've been doing it - use an autofire device or aurofire on the MiSTer and fill the screen with bubbles. Doing this on the board the game doesn't slowdown until much later levels, MiSTer it slows down from level 1 I'm afraid if I 'bubble up'the screen enough. It becomes most prevalent in the early stages on levels 3 and 4 on the MiSTer core, where slowdown is really apparent when the screen is busy. On the board it barely slowdowns at all with a busy screen. It's almost easier to see if you 'bubble up' the screen and then jump with the character, he almost jumps in slow motion on the MiSTer core, but it seems more or less normal on the PCB. Sorry man, I can't see many differences in the newest release at the moment. Thanks as always man, your effort level is always amazing.

LordBarker80, 2023-12-09:
> This is the latest version. Use it with the MRA in it. Comparing with your 10 minute video, I think it builds up close to a 1 second delay after 9'24" minutes. That is roughly 30 microseconds/frame. The old version (v1.3.2) had 85us/frame. So we have more than halved the delay. I need to review the core again to see where the rest of the time may be lost. I hope I don't need to do a new Z80 module next! Everyone in the community has trusted this Z80 module to be accurate so far.
> 
> Let me know your feelings about this release.
> 
> [releases.zip](https://github.com/jotego/jtcores/files/13615819/releases.zip)

Sorry I've replied in a very detailed manner above! In short the intro loop goes out of sync in the same place and this seems almost identical to the previous build I'm afraid. Hope my above post was not too confusing! Thanks as always..

jotego, 2024-01-14:
This version corrects a couple of things in the MCU. The same MCU core is now tested in three cores: JTDD, JTBUBL and JTSHOUSE. So this should be rather robust by now. 

The changes may not have an effect on this issue as maybe the rest of the time difference is no longer due to the MCU. But just in case, please @LordBarker80 please give it a go.

[jtbubl_240114.zip](https://github.com/jotego/jtcores/files/13930933/jtbubl_240114.zip)

LordBarker80, 2024-01-20:
> This version corrects a couple of things in the MCU. The same MCU core is now tested in three cores: JTDD, JTBUBL and JTSHOUSE. So this should be rather robust by now.
>
> The changes may not have an [...]

Thabks for this. Speed still had the same issue. Haven't had a chance to test the intro yet.

LordBarker80, 2024-02-08:
Have tried the intro loops and a similar issue. They start in sync then gradually the real PCB runs ahead until they are out of sync. I have recorded a video that essentially shows the MiSTer core lagging behind the PCB as it good through the attract mode.

jotego, 2025-04-25:
Issue in Rolling Thunder core unveiled a possible cause for the discrepancy today. See commit b1b5e22f2f3224fe7ebbce5768562ebdd4274a8c
To do: check whether the same applies here.

jotego, 2025-04-30:
After testing with `jtsim  -video 1500 -load -w 1400`, the MCU does not use any timer, so the issue solved in b1b5e22 is unrelated to this.

BTW, there was a mismatch in the duration of BSR instruction fixed in 950aa34 which may have an effect here.

real-amano, 2025-12-15:
@LordBarker80: can you check if a current Jotego commit improved something? It's the following commit which could possibly affect Bubble Bobble's timing:  https://github.com/jotego/jtcores/commit/c0102e6f6aa5b29a947b081606fc085fd61f6e72

The PCB is said to use that Motorola M6801 chip.

I don't want to raise any hope though, it's just wishful guessing and not based on any knowledge about those things.

jotego, 2025-12-18:
I built a version that does not use the SDRAM in this [branch](https://github.com/jotego/jtcores/tree/bubl-nosdram). This removes the clock recovery logic. 

Having tested this version on both FPGA and a long sim versus the SDRAM version, there is no difference. The SDRAM version does not see any delay after 13,000 frames compared to the non-SDRAM one. Hence the clock recovery logic is not related to the original issue.

LordBarker80, 2025-12-19:
> I built a [version](https://github.com/jotego/jtcores/tree/bubl-wait) where I can disable the two different wait states present in the Z80 CPUs: the video wait and the shared memory wait. Disabling them makes the CPUs significantly faster.
> 
> We compared the demo video from the arcade with the no-wait version and the current version in JTBIN ([856edaf](https://github.com/jotego/jtcores/commit/856edafa9963e92a5c8640eab9e6c6cb4851530a)). The demo sequence is slightly inconsistent across the three of them from the very beginning in regards to bonus items and slight details such as the bubbles that float around. The no-wait version could be a bit behind on one demo loop just to recover on the next one, without showing a consistent delay/run-ahead build-up. Having increased the CPU speed, one would expect it to run faster, but there are other elements governing the game speed. Apart from the screen frame rate, it looks like the PRN generator (pseudo-random number) plays a noticeable role.
> 
> I think that the PRN generator in this game is designed to be very sensitive to elements we cannot fully control. For instance, the time from the CPU reset signal to the first effective vertical interrupt. This makes a 1:1 comparison of the game play/demo to be inconsistent.
> 
> As with other systems, the comparison must be made at the different components of the system and the interconnection of them. Having reviewed them and even made a new MCU for this core to guarantee cycle accuracy by design, I think this is as accurate as it gets.
> 
> I will not try to match the initial state of the PRN generator to make the boot-up game equal on both a specific PCB sample and the core. The core is open source, so anyone is welcome to debug this further.

@jotego has this version been pushed to updateall? Thanks

LordBarker80, 2025-12-19:
> Here is the file:
> 
> [jtbubl.rbf.zip](https://github.com/user-attachments/files/24255386/jtbubl.rbf.zip)
> 
> Use a keyboard to disable the wait states:
> 
> * Press SHIFT+7 to disable the video wait state
> * Press SHIFT+8 to disable the shared memory wait state
> 
> You will see a white number like `00000011` on the screen when the two wait states are disabled.
> 
> With both disabled, this version may run faster than real hardware on very busy stages.

Hi @jotego I've had a chance to test this new core enabling and disabling the various states with the corresponding keyboard commands. 

I'll refer to both states off as (11) below. 

With either or both of the states disabled the core intro sequence syncs, as far as I tested, 1:1 with the PCB. 

However, there are still speed discrepancies in game with both turned off (11). This actually presents - to my mind - as not how I would expect. 

For example, key moments of slowdown on the PCB are prominent from level 7. When the Maita/Boris attack as the level opens slowdown occurs. Another key moment is level 10, as Monsta and Zen-Chan move the game slows down fairly prominently due to the varied movement patterns of the characters and the volume of enemies. 

With both states turned off (11) on the core, the slowdown on level 7 and level 10 is less pronounced, as if the game is running too fast (as you mention it might, but this doesn't seem to impact the speed of the intro sequence). However, there oddity here is that the slowdown in level 3 and 4 on the core is still prominent. There is no slowdown at all on these levels on the PCB. On the MiSTer core when the screen gets busy, with enemies and bubbles on levels 1-6 there is a range of pronounced slowdown. 

I've done several tests with autofire and the yellow/orange candy which speeds up bubble firing and the slowdown persists. On the PCB there is no slowdown on these levels no matter how much I fill the screen. 

Going back to my earlier point, it seems strange that disabling the states would lessen the slowdown on level 7 but not overly impact the other levels, although admittedly there is an improvement, just less significant. It also seems odd that the intro sequence remains in sync, so this is clearly not getting 'sped up'.

Just to give software emulation a go I tried FBANeo via Retroarch with VRR turned on and also the Arcade Archives version on PS5. As far as I could tell both of those played almost identically to one another and have all slowdown removed, so not accurate to the PCB in that respect. 

If it cannot be improved further re: accuracy I'd suggest if it doesn't impact anything negatively the best compromise is probably the latest core you linked above with the states off (11). This lessens some slowdown on later levels but ultimately means the earlier levels are somewhat more accurate. Also the intro seems to sync 1:1 in these instances.

jotego, 2025-12-23:
@LordBarker80 thanks a lot for the detailed analysis. I have just played the first 20 levels of the current plubic version 856edaf and I cannot really feel any slowdown but I believe that you can corner the game to a condition where it happens.

It makes sense that these changes speed up things a bit. But, I am not comfortable removing those slow down mechanisms because they are present and necessary on the original board. So removing them to fix the slowdown means cancelling an unknown error by creating a known error of the opposite sign. 

The MAME driver for CPS2 actually does something [similar](https://github.com/jotego/jtcores/issues/1295#issuecomment-3687283861). I respect that approach. But I would rather find the real culprit of this slowdown, which is tiny if noticeable.

Let me review things a bit more about this. **Any ideas or extra pair of eyes to compare the core with the schematics are welcome**.

jotego, 2025-12-24:
It's been running for a few hours now, side by side with the previous version. It has built up a delay of about 7". In the last 1h50min, it added about 2", which means that the new version is about 0.03% faster than the older one. I was estimating the delay before to be 0.017% slower than the board, so hopefully this is in sync now during attract mode, which is the easy one to measure.

LordBarker80, 2025-12-28:
While I appreciate the continued effort, this is still not accurate. While I cannot test the exact accuracy in terms of the numbers you are posting, I can notice the slowdown from the first level and the later levels, specifically level 10, now seems to have totally different slowdown from the board. I need to investigate further, but on first play I can notice the difference straight away.  I'll say a final thank you for this, as it appears making this even close to 1:1 isn't possible at the moment. Perhaps in future your time might pick up the original hardware if you feel it important to preservation. Cheers again :)

commit dec9424e (2023-12-06): Fixes bad instructions in jt6801. Fixes #398