jotego/jtcores #404: Playable

jotego, 2023-10-27:


jotego, 2023-10-29:
When the inputs lock up, resetting several times the MCU unlocks them. Although the solution is not stable and may get locked up again soon after.

jotego, 2023-10-29:
commit 35be12d2
Command `jtsim -inputs -video 250 -w 45`

The MCU should be reading port 1 once per frame, but it suddenly stops doing it at frame 163. The memory-mapped cabinet controls are not read anymore either

[image]

jotego, 2023-10-29:
Forcing A6 as the read value for Tri RAM's address 0 restores the MCU automatically. It still hangs up, but it comes back soon after. Still not playable.

```
assign alt_din = xaddr==0 ? 8'ha6 : p_alt_din;
assign bdin    = baddr==0 ? 8'ha6 : p_bdin;
```

jotego, 2023-10-29:
Disabling the timer interrupts on the MCU prevents it from halting.

jotego, 2023-10-30:
The m6801 module (6801_core.sv) is too slow compared to the HD6301. The HD6301 can fetch the next instruction while executing the current one. That makes many implied instructions, like the flag ones (CLC, CLI, TAP...), execute in a single cycle. Branch instructions execute in 3 cycles instead of 4, MUL takes 7 instead of 10, etc. So using m6801 in jt63701v results in a slower MCU, but it is unclear by how much.

Increasing the mcu_cen to 2x makes the game almost playable. The inputs still get stuck for a second occasionally when the MCU is producing PCM sound but input sampling is restored automatically.

jotego, 2023-10-30:
Running the m6801 module at 3x renders the game fully playable, with no stuck inputs.
Commit bdc2f770d78b0c33f3543728cc3f097361777a83

atrac17, 2023-10-30:
I'll ping you about this elsewhere, but I'm going to give the HD63701 you wrote a go in Pac-Land, Sky Kid, Baraduke, Dragon Buster and Metro-Cross by Namco and compare it to the source for the HD63701 with a tb that I received from Mr. X yesterday.

Very curious to see if your implementation fixes the issue of sound playback in Baraduke. Great work as always!

jotego, 2024-01-14:
The cycle-accurate MCU still causes control halts in Splatter House, similar to the old MCU running at 2x. Disabling cycle accuracy in 6301.yaml:

```
config:
  implicit: true
  ignore_cycles: true
```

gives a marginal speed boost, which is enough to make it work smoothly. The culprit is likely to be the IRQ generation as when PCM is busy, the MCU may not see the VB IRQ.

[image]

closed 2024-01-14 by jotego