jotego/jtcores #410: shouse: MCU lock up

jotego, 2023-10-28:
The MCU, sound CPU and other CPUs in Namco System 1 communicate through the _TRIRAM_. MAME uses a hack to get the system to boot by locking writes from the MCU to _TRIRAM_'s address zero after the value $A6 has been written (see namcos1_m.cpp line 584). This does not match the schematics as the MCU is always allowed to write anywhere. On the FPGA core, it is also required to boot up, but it shouldn't.

Things to look at:

- The MCU writes to the EEROM chip and then waits until it reads the same value from it. This process must have taken a few checks on a real EEROM, slowing down the MCU
- The mapper model may be wrong, particularly about the reset signal to the sub, MCU and sound CPUs

jotego, 2023-10-29:
**Introducing a delay after each MCU write to EEROM**

- Forcing a delay after EEROM writings equivalent to 5 check-loop iterations in the MCU do not solve the lock-up issue
[image]

-

jotego, 2023-10-29:
The reset signal and LVBL are not related in the PCB. But, the PCB does lock up during boot up occasionally.

The core around commit 82ccf09ca0ca798f3048c502918a8c0b9b923a36 may fail to boot up in simulation when bypassing the ROM load. Comparing waveforms shows that the timing of the VB-related IRQ is important. Linking the SUBRES to the VB edge fixes the problem in simulation. Still, the sound and MCU are messed up during game play.

**PCB measurements of LVBL and global RESET**
Most consistent relationship after a while with no power:
[image]

Driving down the supply not to zero and then back up again shows other relationships
[image]
[image]

gyurco, 2024-03-22:
The $a6 hack breaks Shadowland. The game does a RAM check, and as it reads back $a6 from $1000 (in the CPU's POV) instead of the value it written before, it fails. But even removing the hack, this game doesn't want to work, it stops later on (the memtest passes at least).

gyurco, 2024-03-23:
This one works with all games so far:
https://github.com/jotego/jtcores/commit/eea4c381234b824a7e2b0a7b42bed0ee02eacab2

jotego, 2024-03-23:
Very interesting to see it working well with a different approach. It still does not make sense that all this is needed from looking at the schematics, but I guess we'll have to live with the hack. 
My PCB does not want to boot anymore and debugging a PCB can be more work than designing a core so I may not be able to research further into this odd lock up.

closed 2024-03-23 by jotego