jotego/jtcores #411: shouse: verificar todos los volcados de ROM de programa

jotego, 2023-10-29:
El _driver_ de MAME para Namco System 1 tiene un apaño que no tiene sentido ni para el autor. Podría ser necesario porque el volcado de las ROM es incorrecto.

Verificar todas las ROM de código de los tres procesadores 6809

- [ ] sh1_prg0.bin
- [ ] sh1_prg1.bin
- [ ] sh1_prg2.bin
- [ ] sh1_prg3.bin
- [ ] sh1_prg4.bin
- [ ] sh1_prg5.bin
- [ ] sh1_prg6.bin
- [ ] sh1_prg7.bin
- [ ] sh1_snd0.bin
- [ ] sh1_snd1.bin

etriana85, 2023-12-05:
Volcado disponible en "nobackup"

jotego, 2023-12-07:
El resultado de la comparación es que tres de las ROM de programa son distintas. Es necesario usarlas para ver si el problema de arranque de la MCU desaparece así.

etriana85, 2023-12-11:
ROM_START/REGION|   shouse dif.   |splatter      |splatter 2              |splatter j                  |
-----------------|-----------------|-----------------|-----------------------|--------------------------|
sh1_prg0.bin|a0da1a24	|    _**4e07e6d9**_  |	a0da1a24	|a0da1a24	|
sh1_prg1.bin|7a3efe09	|7a3efe09	|7a3efe09	|7a3efe09	|
sh1_prg2.bin|434dbe7d	|434dbe7d	|434dbe7d	|434dbe7d	|
sh1_prg3.bin |955ce93f	|955ce93f	|955ce93f	|955ce93f	
**_sh1_prg4.bin_**|**_dfbe0e57_**	|350dee5b     |350dee5b|      350dee5b                   |
sh1_prg5.bin|0187de9a	|   _**3af893d9**_        |0187de9a	|0187de9a	
**_sh1_prg6.bin_**|**_bf5b20eb_**	|_**0579e90e**_|  _**054d6275**_         |  _**7a3e664**_                      |
sh1_prg7.bin|24c8cbd7	|    _**653b4509**_      |     **_942cb61e_**  |24c8cbd7	|
sh1_snd0.bin|90abd4ad	|    **_03b47a5c_**       |90abd4ad	|90abd4ad	
sh1_snd1.bin|8ece9e0a | 8ece9e0a | 8ece9e0a| 8ece9e0a|