jotego/jtcores #418: Sangokushi II (Asia 921005).mra background image wrong.

mhmetalheart, 2023-11-02:
Sangokushi II (Asia 921005).mra, wofa.zip. 
I checked Tenchi wo Kurau II Sekiheki no Tatakai (CPS Changer, Japan 921031).mra (wofch.zip). Found some different. 
See line 39 to 41 from wofch:
           32 FF 00 02 04 06 26 28
            2A 2C 2E 36 30 30 02 04
            08 30 25 40 88 73 FF 20
from wofa:
            FF FF FF FF FF FF 22 24
            26 28 2A 00 00 2C 10 08
            04 00 25 40 88 73 FF 20

After replaced by the wofch. The game background image shown. But still wrong, have some vertical lines which is totally wrong. I cannot fix them. Please help.

And, have a request. I download hbmame wof.zip. (HBMAME 0.199), use sk2h23 made a mra. Which is a Chinese bootleg wof, called SAN JIAN SHENG. I almost done everything except the title logo. It show 三国志II （same as asia ver.) Which actually should be 三剑圣. I don't know the rom included the logo, or not. If included, the logo in different address? If not, that means I use wrong game version? Anyway, my friends happy that it worked. If you haven't any time to make one. Just ignore my request.

jotego, 2024-01-20:
Thank you for bringing this up. I had not noticed that some _wof_ versions had the wrong mapper.

Your problem with the logo is likely to be related to game settings (press F2 to enter the test menu or look in the DIP switches) or to using the wrong ROM for the backgroun data (most likely this one).

closed 2024-01-20 by jotego

mhmetalheart, 2024-02-28:
Hello, the last 2 updates resolved the Sangokushi II Asia version's issue. That's great. Thanks for hard job. (though I already wrote one for me and my friends ago. But still thanks! We're very happy you see the issue and resolved it.)
I have a request, can you add the mra for: wofabl? Sangokushi II asia Chinese bootleg.
I remember you wrote Sangokushi II huo feng huang (wofhfh) ago, it works great. But seems you deleted. I try to modify wofabl based on this one. But always failed. I'm too weak. : )
We Chinese all love these old time which played on Arcade. Hope you can bring them back. 
These roms cotained in mame 0.255 merged.

commit 8bd50e33 (2024-01-20): Warriors of Fate: some versions did not have the right mapper. Fixes #418