jotego/jtcores #419: shouse: PCB gameplay comparison with beta

jotego, 2023-11-02:
By _owlnonymous_:

> The main issues I witnessed (which are probably already reported):
-Samples sound correct but music sometimes chooses wrong instrument, speed, or pitch. Sometimes after restarting a level the music sounds more correct
-Some items spawning in higher than normal. Examples are the rocks you can pickup on stage 3, the items that fall from the chandelier, and the boreworms from the pit you can fall in.
-Some screen tearing on the outside left and right side
-Lack of lag in enemy heavy areas
>
>  https://youtu.be/HuPG688x-Ms

jtmiki, 2024-04-11:
I watched the part of Splatter house of the video and the differences between JT cores and the original PCB are as follows;

- **The 1st and the 3rd stage**  Sound effects are a little bit lower.
- **The 3rd stage** The slide move is just when you're about to land you hit down and attack.  It's one of the things that's really flawed in MiSTer the timing is really off.  It feels a lot different on the PCB board feels a lot better on the PCB board. 
- **The 7th stage** That's twice that the Rocks disappeard to completely not come down at all which is unusual worth nothing I don't think it's an error in the core though just a weird RNG.

He adjusts screen settings when he plays JT cores (MiSTer) one. 
| CRT | NUM |
|-----|------|
| H offset | -8 |
| V offset |  0 |
| scale enable | off |

jotego, 2024-06-26:
Currently none of the PCBs we own boot up anymore :(

closed 2024-06-26 by jotego