jotego/jtcores #425: TMNT: TMNT Metallic noise in some early scenes

jtmiki, 2023-11-09:
TMNT
Metallic noise in some early scenes:
-[   ] **When the letters appear announcing the second stage**
-[   ] **During character selection scene**
[Player_Selection_MISTER TMNT.zip](https://github.com/jotego/jtcores/files/13306420/Player_Selection_MISTER.TMNT.zip)
[Player_Selection_MAME TMNT.zip](https://github.com/jotego/jtcores/files/13306424/Player_Selection_MAME.TMNT.zip)

Commit: fd8f15e

jotego, 2023-11-12:
There is a bit of clipping on some sounds. There is no clipping during the second stage title screen, but the sound is awkward

jotego, 2023-11-12:
The sound at the second stage noise occurs while the player's energy is charged up. Using https://github.com/jotego/jt51/commit/898faf289488acf5a7d12df9062a814966b75158 works fine:
[898faf2.mp3.zip](https://github.com/jotego/jtcores/files/13327509/898faf2.mp3.zip)

But version https://github.com/jotego/jt51/commit/a9907cd46ebd6a4830d2ae1bd9600f87d5396f92 breaks it:

[a9907cd.mp3.zip](https://github.com/jotego/jtcores/files/13327521/a9907cd.mp3.zip)

commit f57e7783 (2023-11-12): Fixes #425
closed 2023-11-12 by jotego