jotego/jtcores #431: Bugs of BUBL, KIWI, SBASKT and ST(POCKET)

jtmiki, 2023-11-16:
There is a bug of each core.

**BUBL**: Immediately after the start of first stage, the screen goes black and after a few seconds the NOTICE screen appears.
**KIWI**: The screen is black and none of the games start.
**SBASKT**: Screen orientation is upside down.
**SF**: In the first stage, the sound effects when the protagonist attacks are not timed correctly.

Commit: 57155bcc40677c679a453dbee04527e4a97eb8a4

jotego, 2023-11-16:
- KIWI: broken in commit 1839fea0285b1a1d629cd89f0a24a0856b63e5b4
- BUBL: broken with the current version of the MCU. See #432. Reverting to previous core version until then
- SBASKT: an old file was in the common folder rather than in _alternatives. Deleting it fixes the problem
- SF: the game is just that way

jotego, 2023-11-17:
KIWI issue moved to https://github.com/jotego/jtcores/issues/433
Using old RBF for today's update

closed 2023-11-17 by jotego