jotego/jtcores #437: Street Fighter II Champion Edition (Red Wave) 1, 2 missing

0xPerseus, 2023-11-18:
Looks like these were removed in the following commit. 

https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d

Would be great to have both of these added back if possible.

Street Fighter II' Champion Edition (Red Wave PtII, bootleg).mra
Street Fighter II' Champion Edition (Red Wave, bootleg).mra

Thank you

ksdomino, 2023-12-21:
These were excellent variants that were in local arcades all over the place a year after Street Fighter 2 was around. I think they got deleted by accident as dupes but they are a part of the Street Fighter History (and the reason we got the official turbo streetfighters -not as good as these) so and would also love to see these preserved too. The RE/Rainbow Editions got deleted too, if fact ALL these (some are dupes) :

[...' Champion Edition/Street Fighter II' Champion Edition -Accelerator Pt.II, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-d86a38e7da420c650de98a35eeca146345a7f16d16ecf76e1ee5c88e95046d1f)
This file was deleted.

[... Champion Edition/Street Fighter II' Champion Edition -Accelerator!, bootleg, set 1-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-05317780f96a4cf36372d756ef8b66538b616ce66ec91681fa0f3578f786fe7e)
This file was deleted.

[... Champion Edition/Street Fighter II' Champion Edition -Accelerator!, bootleg, set 2-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-eb4680b6de368cc567bb92c3020d9803176cce149af99fe6c8f954edd0a0fe40)
This file was deleted.

[...ghter II' Champion Edition/Street Fighter II' Champion Edition -Dstreet-6, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-521a1297f3da8632e73eaddfd56b5b19df87574013674f5663f747c573ef0457)
This file was deleted.

[...eet Fighter II' Champion Edition/Street Fighter II' Champion Edition -M10, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-b7a2529f337fb2bd1dbf36f701f30809a713ae94e45fb5611be3d23a8281790a)
This file was deleted.

[...reet Fighter II' Champion Edition/Street Fighter II' Champion Edition -M3, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-fa27a5faa9af9c7813ab674a22b1c43ac49aae35ed9d2185f01261640e038988)
This file was deleted.

[...I' Champion Edition/Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 1-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-e3948584c5d499748a73d9c91e305bf363ff018d5e8eef0d44a7d60137fd145f)
This file was deleted.

[...I' Champion Edition/Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 2-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-1f853ca0695907c905dc871b053b83624ec20551abaad6d06371a6bea4ecf953)
This file was deleted.

[...I' Champion Edition/Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 3-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-adbd0515d610f54cb50e53690f21e7f3433f394defba0a1bc53682ddd5a920dd)
This file was deleted.

[...reet Fighter II' Champion Edition/Street Fighter II' Champion Edition -RE, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-2b042f93352aaaff420aceb2969433682f1dfbd3be9aab8390d56978f45a4deb)
This file was deleted.

[... II' Champion Edition/Street Fighter II' Champion Edition -Rainbow, bootleg, set 3-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-b5f969c02b25627716a18e8df4f4cfe5dfeb494640ae6b933039e2375f57a436)
This file was deleted.

[...r II' Champion Edition/Street Fighter II' Champion Edition -Red Wave PtII, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-d766ada946d66511825875cb589cfdcd40d3bf01e595deb7a1462ecff9390f15)
This file was deleted.

[...ighter II' Champion Edition/Street Fighter II' Champion Edition -Red Wave, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-0d015bd6aeeccc79a7effd427c1f251ee89563f0db6da4caba55a76a57106154)
This file was deleted.

[...reet Fighter II' Champion Edition/Street Fighter II' Champion Edition -UPL bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-662d274eb7c21d0a0af263fcde6f2c7da45d0f060938ad94a9bd51f0a5559f60)
This file was deleted.

[...et Fighter II' Champion Edition/Street Fighter II' Champion Edition -V004, bootleg-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-524c6a6d50f95559cd67685b9e4acb8c949aca8893b7a439036b5293e0135929)
This file was deleted.

[... Champion Edition/Street Fighter II' Champion Edition -bootleg with level selection-.mra](https://github.com/jotego/jtbin/commit/562d6950a2789190162a5116bac0fab8cbf59c4d#diff-c5edabb79cd2b30c1c8fdff3a322ec1737006e7380fcd44c6da2231211d3b877)
This file was deleted.

jotego, 2024-01-14:
The following games will be reinstated next Friday:

- Pang! 3 (bootleg, set 1).mra
- Street Fighter II' Champion Edition (Accelerator!, bootleg, set 1).mra
- Street Fighter II' Champion Edition (Accelerator!, bootleg, set 2).mra
- Street Fighter II' Champion Edition (Accelerator Pt.II, bootleg).mra
- Street Fighter II' Champion Edition (Rainbow, bootleg, set 3).mra
- Street Fighter II' Champion Edition (Red Wave, bootleg, set 1).mra
- Street Fighter II' Champion Edition (Red Wave PtII, bootleg).mra
- Street Fighter II' Champion Edition ('Taiwan' bootleg with PAL).mra
- Street Fighter II' Champion Edition (UPL bootleg).mra
- Street Fighter II' Champion Edition (V004, bootleg).mra
- Street Fighter II' Champion Edition (YYC, bootleg).mra
- Street Fighter II The World Warrior (Quicken, bootleg).mra
- Street Fighter II The World Warrior (Quicken Pt-I, bootleg).mra

commit 9eda3d6e (2024-01-14): reinstates 13 bootleg games for JTCPS1. Fixes #437
closed 2024-01-14 by jotego