jotego/jtcores #446: PCBs needed for schematic extraction

jotego, 2023-12-05:
Taito games using the 68705 MCU:

- [ ] 40 Love (TAITO)
- [ ] Birdie King (TAITO)
- [ ] Birdie King 3 (TAITO)
- [ ] The Legend of Kage (TAITO)
- [ ] LSA Squad (TAITO)
- [ ] Mat Mania (TAITO)
- [ ] NY Captor (TAITO)
- [ ] Pit & Run (TAITO)
- [ ] Return of the Invaders (TAITO)
- [ ] Slap Fight (TAITO)
- [ ] Xain'd Sleena (TAITO)

etriana85, 2024-01-02:
He encontrado solo los manuales de:
- The Legend of Kage (TAITO)
- LSA Squad (TAITO)
- Mat Mania (TAITO)
- NY Captor (TAITO)
-  Slap Fight (TAITO) - A.L.C.O.N

- Birdie King 2 (TAITO): Descargué el sch, pero aparecía Birdie King 2, no 1 ni 3

jtmiki, 2024-02-08:
I bought a NEW Taito F3 mother board and 2 cartridges.
We will get them next week.

closed 2024-03-22 by jotego

jtmiki, 2024-03-25:
I bought a PCB of Dragon Ball Z and it arrived yesterday.