jotego/jtcores #447: Renegade (PCB): volcar memorias

jotego, 2023-12-05:
Volcar las memorias de la placa de Renegade que tenemos. Esta placa es peculiar porque tiene MCU pero no tiene marcas de TAITO por ningún sitio. La ROM podría ser diferente.

etriana85, 2023-12-05:
He volcado las ROM de maincpu, audiocpu, adpcm, tiles, sprites y chars. 
He verificado el CRC de MAME vs Checksum, y no coincide ninguno. 
El volcado esta disponible en nobackup bajo el folder Renegade.

jotego, 2023-12-06:
Puede que la grabadora de EPROMs no calcule el CRC del mismo modo que MAME. Hay que recalcular el CRC usando el comando `crc32` en el terminal de linux.

etriana85, 2023-12-07:
Después de recalcular el CRC según instrucciones, se encontró que en la PCB que tenemos, no coincide el CRC de maincpu con respecto a los datos de mame.

Juego           |   maincpu  |  
-----------------|----------------|
R Diferente  |    fd09e9ef |
Renegade   |    ba683ddf|
Renegadeb |    3dbaac11|
Kuniokun     |   93fcfdf5   |
Kuniob         |   a90cf44a |

closed 2023-12-07 by etriana85