jotego/jtcores #449: kiwi: sound effects are wrong (MISTER)

jtmiki, 2023-12-07:
The New Zeland Story

A bad sound effect is played when using the fireball baton weapon. 
In the Service Test of the game it can be played with "SOUND 16".
commit: b25223e

jotego, 2024-01-14:
fixed in 009d746

skutis, 2024-01-14:
I think the reference https://github.com/jotego/jtcores/commit/009d7461bf035e35dcab0f6d68e01732fbee7769 is wrong here.

jotego, 2024-01-18:
it is still broken in 415c3d2

jotego, 2024-01-21:
Reproduce with 

- `jtsim -setname tnzso -dipsw 0xfffa -video 360 -w 300 -inputs -d VERILATOR_KEEP_JT03`
- [sim_inputs.hex.zip](https://github.com/jotego/jtcores/files/14000509/sim_inputs.hex.zip)

jotego, 2024-01-21:
The CPU is writting this to JT49:

[image]

This is the data written by the equivalent [VGM file](https://github.com/jotego/jtcores/files/14000657/tnzs_0.vgm.zip)  from MAME

[image]

There seems to be a system-level problem. BTW, disabling the recovery for the Z80 prevents the system from booting up.

closed 2024-01-21 by jotego