jotego/jtcores #453: Outrun on AP - "Error in framework. Core won't enter reset"

epbarger, 2023-12-11:
Running Outrun on analogue pocket firmware 1.2, when quitting the core I see the message "Error in framework. Core won't enter reset" about 50% of the time. After this error, I'm more likely to experience corruption/freezing issues on other cores, which otherwise have no issues when booted before this error from Outrun.

jotego, 2023-12-19:
Do you only see that in JTOUTRUN or does it happen in other JT cores too?

epbarger, 2023-12-19:
I have only experienced the issue on Outrun, but that is the core I use the most by far. I will comment if I experience it on other cores

ardaniel, 2023-12-23:
I've seen this issue in JTSF on the AP once, also on firmware 1.2. (I got my AP that same week so never had a chance to test on old FW.)

jotego, 2024-09-19:
I was not able to reproduce this and I have not hear user complaints about it in a long time. I suppose this works now. Reopen the issue if you still see it in the recent updates.

closed 2024-09-19 by jotego