jotego/jtcores #455: Graphical glithches with bubble Bobble core (pocket)

mpatour, 2023-12-12:
Hi
As I reported on Patreon, I've encountered some graphical issues after the last update of Bubble Bobble core for the Analogue Pocket. Thoses glitches happen with Bubble Bobble and Scamble.
[image]
[image]
[image]
[image]
[image]

jotego, 2023-12-12:
I think this is not the core's fault. I was expecting some marginal glitch in an advanced level. These glitches are very obvious and we would have caught them in our testing straight away.
I think there is something else going on here. Maybe the .rom files are not correct? The MRA files changed in the latest version and that required new .rom files generated. Make sure you have .rom and .rbf_r (core) files matched on the same version.

jotego, 2023-12-22:
Sorry, you were right. This is specific to the Pocket version. The MiST version works ok, and that's the main one used for testing. I need to dig a bit into it to find out why the core gets this bug when compiled for the Pocket.

stilltruckin, 2023-12-24:
Hi, I have the same graphic glitch with jtbubl core as pictured above.

jotego, 2024-01-15:
- not related to the jtframe_dual_ram Pocket-specific code
- Only happens to the scroll layers, characters and objects are visible
- Tokio (M6805) affected too
- Only Pocket, MiST and simulation are fine
- A tile may be visible in the affected area and move with along the scroll

[image]

jotego, 2024-01-16:
The graphics decoder PROM data (u_gfx.u_prom) is not loaded correctly in the Analogue Pocket.

The address does not get past FD, missing the last two bytes: FE, FF. Taking the header into account, the last 4 bytes are missing.

The problem comes from the **Analogue Pocket firmware,** which **does not handle file transfers correctly if the file size is not multiple of four.**

[image]

commit 08c5488a (2024-01-16): Fixes #455
closed 2024-01-16 by jotego