jotego/jtcores #457: How to disable the debug overlay

gyurco, 2023-12-13:
Hi!

I recompiled almost all jtcores, but I noticed the debug overlay is always displayed.
- I used -u JTFRAME_DEBUG
- Checked the resulting QSF doesn't have JTFRAME_DEBUG
- The core(s) still have the debug overlay at the bottom

```
grep VERILOG_MACRO jtaliens.qsf 
set_global_assignment -name VERILOG_MACRO "NO_DIRECT_UPLOAD=1"
#set_global_assignment -name VERILOG_MACRO "USE_QSPI=1"
set_global_assignment -name VERILOG_MACRO "DUAL_SDRAM=1"
set_global_assignment -name VERILOG_MACRO "I2S_AUDIO=1"
set_global_assignment -name VERILOG_MACRO "VGA_8BIT=1"
set_global_assignment -name VERILOG_MACRO "USE_HDMI=1"
set_global_assignment -name VERILOG_MACRO "MIST=1"
set_global_assignment -name VERILOG_MACRO "ALIENS="
set_global_assignment -name VERILOG_MACRO "COMMIT=52a708db-dirty"
set_global_assignment -name VERILOG_MACRO "CORENAME=JTALIENS"
set_global_assignment -name VERILOG_MACRO "CORE_OSD="
set_global_assignment -name VERILOG_MACRO "DATE=231213"
set_global_assignment -name VERILOG_MACRO "GAMETOP=jtaliens_game_sdram"
set_global_assignment -name VERILOG_MACRO "JTFRAME_4PLAYERS=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_ARX=4"
set_global_assignment -name VERILOG_MACRO "JTFRAME_ARY=3"
set_global_assignment -name VERILOG_MACRO "JTFRAME_BA1_START='h30000"
set_global_assignment -name VERILOG_MACRO "JTFRAME_BA2_START='hB8000"
set_global_assignment -name VERILOG_MACRO "JTFRAME_BA3_START='h240000"
set_global_assignment -name VERILOG_MACRO "JTFRAME_BUTTONS=3"
set_global_assignment -name VERILOG_MACRO "JTFRAME_COLORW=8"
set_global_assignment -name VERILOG_MACRO "JTFRAME_COMMIT=32'h52a708d"
set_global_assignment -name VERILOG_MACRO "JTFRAME_CREDITS=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_CREDITS_HSTART=32"
set_global_assignment -name VERILOG_MACRO "JTFRAME_CREDITS_PAGES=4"
set_global_assignment -name VERILOG_MACRO "JTFRAME_DIALEMU_LEFT=5"
set_global_assignment -name VERILOG_MACRO "JTFRAME_DIPBASE=8"
set_global_assignment -name VERILOG_MACRO "JTFRAME_HEADER=8"
set_global_assignment -name VERILOG_MACRO "JTFRAME_HEIGHT=9'd224"
set_global_assignment -name VERILOG_MACRO "JTFRAME_INPUT_RECORD=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_INPUT_RECORD_AW=12"
set_global_assignment -name VERILOG_MACRO "JTFRAME_IOCTL_RD=4096"
set_global_assignment -name VERILOG_MACRO "JTFRAME_JOY_DURL=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_MCLK=48000"
set_global_assignment -name VERILOG_MACRO "JTFRAME_MEMGEN="
set_global_assignment -name VERILOG_MACRO "JTFRAME_OSD_VOL=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_PLLSIM=10.416"
set_global_assignment -name VERILOG_MACRO "JTFRAME_PROM_START='h3C0000"
set_global_assignment -name VERILOG_MACRO "JTFRAME_PXLCLK=6"
set_global_assignment -name VERILOG_MACRO "JTFRAME_SDRAM_BANKS="
set_global_assignment -name VERILOG_MACRO "JTFRAME_SIGNED_SND=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_TIMESTAMP=1702462517"
set_global_assignment -name VERILOG_MACRO "JTFRAME_VERTICAL=1"
set_global_assignment -name VERILOG_MACRO "JTFRAME_WIDTH=9'd320"
set_global_assignment -name VERILOG_MACRO "PCM_START='h38000"
set_global_assignment -name VERILOG_MACRO "SEPARATOR=-;"
set_global_assignment -name VERILOG_MACRO "SIDI128=1"
set_global_assignment -name VERILOG_MACRO "TARGET=sidi128"
```

jotego, 2023-12-19:
Compile with `jtcore --nodbg` or define `JTFRAME_RELEASE`. Using `--git` will also remove it and place the output files in whatever folder is pointed to by the `$JTBIN` environment variable.

jotego, 2023-12-19:
By the way, `JTFRAME_DEBUG` was deprecated a few months back. The default behavior now is to show debug features unless `JTFRAME_RELEASE` is defined.

closed 2023-12-19 by jotego

gyurco, 2023-12-19:
Good to know, thanks.
Can you look at https://github.com/jotego/jtcores/pull/456 please?