jotego/jtcores #458: s16: Tetris - corrupted background image

Thelypody, 2023-12-14:
Tetris (Set 4, Japan, S16A) [FD1094 317-0093]. Playing on MiSTer, Main 20231018, jts16 1.2.46.

One particular background image appears to miss some of its graphic tiles, leaving a very noticeable hole to the left of playfield.
[image]

jotego, 2023-12-19:
This looks like a ROM issue. Could you try that ROM set on MAME?

Thelypody, 2023-12-19:
I've tested the same rom set file on a fresh install of MAME 0.261. The emulator had accepted the files and eventually the same bug was encountered. Level 6 background is corrupted.
[image]
_Set 4, MAME._

I dug a bit and found some possible clues.

1. [The same bug was discovered in MAME back in 2008.](https://mametesters.org/view.php?id=2030) It is still listed as open to this day.
2. There are four official sets of Tetris for System 16 arcade boards. Sets 1 and 2 are listed as belonging to System 16B, and sets 3 and 4 to System 16A. I've tested them all on MiSTer and found that the bug in not present in sets 1 and 2, while set 3 shows an identical result to set 4.
[image]
_Set 1, System 16B_
[image]
_Set 2, System 16B_
[image]
_Set 3, System 16A_

At this point I'm no longer sure if that's an emulation bug or a real hardware behavior. Given how obvious the bug would have been during testing and the fact that MAME still lists it as a bug makes me lean towards being an emulation issue.

jotego, 2023-12-19:
I think the dumped ROM may be corrupted. The way the GFX hardware works makes this bug unlikely to be hardware related.
The GFX basically have a large video RAM of which you see a portion on screen. But that portion does not break in the middle of a row in the screen like in the image.
Thank you for enquiry. I will close it for now as I think this is a ROM problem.

closed 2023-12-19 by jotego