jotego/jtcores #459: Analogue Pocket Gun Smoke: All ROMs Upside Down Except "Gun.Smoke (World, 1985-11-15)"

bebopjp, 2023-12-14:
Using jtgunsmk 1.5.0 on the Analogue Pocket (OS v1.2), all the ROMs pulled by Pocket Updater are upside down, except "Gun.Smoke (World, 1985-11-15)".

I noticed in some other JTCores running on Analogue Pocket, there is a "Flip Screen" option.  But in jtgunsmk, I can only see a "Rotate Screen" option.

nortakales, 2023-12-24:
Simpsons is also upside down, with no core option to flip/rotate the screen

jotego, 2023-12-25:
> The Simpsons (4 players World, set 1) is also upside down, with no core option to flip/rotate the screen. However, set 2 is NOT upside down. I did not check further ROMs.

Your system probably has an old and corrupted save file from an earlier version of the JTSIMSON core. Delete it and it will work.

jotego, 2024-01-14:
About Gun Smoke, everything seems fine in commit f8166fbb. It must have been fine for a long time. I will update the files in JTBIN next Friday just in case that copy was bad.

closed 2024-01-14 by jotego