jotego/jtcores #460: JT74 KiCAD incorrectly inverted pin on LS241

antongale, 2023-12-15:
Pin 19 should not be inverted on the 74LS241

change this:

      (pin input inverted (at -12.7 -12.7 0) (length 5.08)
        (name "OEb" (effects (font (size 1.27 1.27))))
        (number "19" (effects (font (size 1.27 1.27))))
      )

to this:

      (pin input line (at -12.7 -12.7 0) (length 5.08)
        (name "OEb" (effects (font (size 1.27 1.27))))
        (number "19" (effects (font (size 1.27 1.27))))
      )

jotego, 2023-12-19:
@etriana85 please check pin 19 of JT74's LS241

etriana85, 2023-12-19:
I have checked the 74LS241 device datasheet, and pin 19 is active high. 
I have corrected the symbol in KiCad. Please pull to update the changes. 
[image]
[image]

closed 2023-12-19 by etriana85