jotego/jtcores #463: Requisitos de fabricación  PCBWay

etriana85, 2023-12-20:
- [ ] Fresado: 0,2 ~ 0,5mm
- [ ] Ranurado: 0,5mm
- [ ] Grosor PCB: 0,2~2,4mm
- [ ] Espesor tolerancia: 1,0mm
- [ ] Grosor tolerancia: 1,0mm
- [ ] Distancia entre pistas mínimo:  0,1~4mm (Recomendado 0.15mm min)
- [ ] Espesor de Cu: 1,2,3 oz
- [ ] Espesor capa interna: 1~1,50 oz
- [ ] Perforación: 0,2~6,3mm
- [ ] Anchura anillo anular: 0,15mm
- [ ] Pads con vías en el centro, anchura: min 0,15mm
- [ ] Diámetro del agujero acabado: 0,2~6,2mm
- [ ] Serigrafía min 0,15 x 0,8 mm(H); Relación ancho alto 1:5
- [ ] Agujeros chapados: 0,6mm
- [ ] Mascara soldadura: verde, rojo, amarillo, azul
- [ ] Serigrafía: blanco, negro

jotego, 2024-01-01:
- Mover esto a jtkicad/doc

etriana85, 2024-01-02:
Información archivada en jtcores/modules/jtkicad/doc/pcbway.md

closed 2024-01-02 by etriana85