jotego/jtcores #474: rom_ok is flawed

gyurco, 2024-01-04:
At the latest git, building yiear (and probably other similar cores) result in a broken core. The problem is that the rom_ok signal comes earlier before the valid data, makes the CPU executing bogus instructions. Check this capture:
[image]

At 8164h, the correct byte would be CEh, but the rom_ok arrives one cycle sooner (even 8163h is wrong, because 10h is the correct value, but at least the cen_E was delayed to compensate).

Reproducing it is easy, usually it fails somewhere at the first 5-10 instructions (here failed at the first).

jotego, 2024-01-06:
I have not been able to reproduce a bad_ok in simulation. The problem actually came from a regression error introduced in commit 022496cbf4399b9348b4af9ab01ab26e587c6ee0. The default value for a parameter was changed and the file committed by mistake. Commit d0553e9 reverts the change.

Under that change the CPU of several games ran at 4x the speed. 

Thank you for reporting this, I had not noticed it.

closed 2024-01-06 by jotego

gyurco, 2024-01-06:
I confirm it's working now, thanks!
P.S. I wonder if it works because there's no problem with rom_ok, or just the normal speed CPU doesn't care.

jotego, 2024-01-07:
I think that if rom_ok was bad, all cores would fail constantly. Your signal tap does not compare rom_data with rom_ok, but the cpu_din signal, which changes regardless of rom_ok. I think the rom_ok was a red herring in this case.

gyurco, 2024-01-07:
Yes, but cpu_din should be right, when cen_E is high, regardless of the CPU speed, and here's not. But maybe there's something in the middle I'm not aware of.

commit d0553e93 (2024-01-06): Fixes #474