jotego/jtcores #478: bubl: question about EXTEND

jotego, 2024-01-07:
The EXTEND letters are supposed to appear in a certain order according to [Nicola Salmoria](https://mamelife.blogspot.com/2006/08/completed-at-last.html):

- Either ETN or XED

lord_dOt reported via Discord that he has seen out of sequence letters. I have not been able to reproduce the problem in v1.5.5.

_Note:_ the number of letter bubbles depend on how many enemies you killed at once in the previous stage.

lorddOt7, 2024-01-08:
The issue is easily replicated by waiting 2 complete loop of the attrac mode (after a cold boot of the mister) and by starting a game the third time the bubble bobble logo appear on the screen. On the third or fourth level the letter appear not based on the two block ETN or XED. 
[image]
[image]

lorddOt7, 2024-01-08:
Tried custom version 03c92221 with no luck. The bug remains.

jotego, 2024-01-09:
Thank you for taking the time to look into this. I will close the issue then.

closed 2024-01-09 by jotego