jotego/jtcores #480: Convert to mem.yaml format

jotego, 2024-01-07:
The following cores still have the logic to download the data to the SDRAM and access it in the _game_ module. Take it out from there by creating a mem.yaml file, like in the rest of the cores.

For each core:

- Remove any module like `jt1943_prom_we` and `jtframe_rom`
- Assign the memory signals to SDRAM banks in a new `mem.yaml` file
- Move game-level `jtframe_prom` file to the BRAM section of `mem.yaml` (1943)
- Delete or simplify core-specific _sdram_ modules such as `jtcop_sdram`

Many cores, like 1942, have undergone this transformation so the git history shows how to do it.

List of cores:

- [ ] 1943
- [ ] btiger
- [ ] cop
- [ ] cps1
- [ ] cps15
- [ ] cps2
- [ ] exed
- [ ] gng
- [ ] gunsmk
- [ ] kchamp
- [ ] ktiger
- [ ] midres
- [ ] ninja
- [ ] pang
- [ ] rastan
- [ ] rumble
- [ ] sarms
- [ ] sectnz
- [ ] sf
- [ ] slyspy
- [ ] trojan
- [ ] vigil

somhi, 2024-01-17:
1943 moving to mem.yaml wip  [commit](https://github.com/jotego/jtcores/compare/master...somhi:jtcores:480)

Empezando a mirarme la migración de cores a mem.yaml. Dudas iniciales:
* JTFRAME_SDRAM_BANKS entiendo se debe poner siempre en macros.def
* No he sabido identificar que usase más de 1 banco por lo que solo uso 1
* Todos los wire prom_xx_we =  hay que inicializarlos y a que valor ?
* No me queda claro que valores poner en la sección download en mem.yaml

si prefieres en inglés contesta directamente en inglés y seguiremos así.

jotego, 2024-01-18:
Let's try in English. If you find it troublesome we can switch to Spanish.

- JTFRAME_SDRAM_BANKS is added automatically, don't do anything.
- the original code only uses one bank, you can leave it like that for now.

The tough part here is moving the **jt1943_prom_we** module to mem.yaml and the game. That module did these functions:

1. collect the ROM download sent by the firmware through ioctl_wr
2. reorder some address signals so that the data is written in a different order than it comes in
3. for the final part of the rom, where the PROMs go, it identifies the different _files_ in it and signals to each jtframe_prom module when to take data in

When you move to mem.yaml, number 1 comes for free, but you have to specify number 2. There are several ways of doing it. Checkout the Confluence documentation. If you do not do it right, the game will normally start but graphics will be corrupted. 

For the PROMs, you basically move the code to the game module. Check out how these signals are done in other cores that have _mem.yaml_

somhi, 2024-01-22:
getting there [commit](https://github.com/jotego/jtcores/compare/master...somhi:jtcores:480) but haven't got the clue yet about reordering address signals.   Need to know a bit more about how that is working.   

[image]

jotego, 2024-01-23:
Good progress!

For the sound just test that the sound CPU starts up correctly. That will tell you that the ROM is loaded right. In order to do that, simulate with sound (do not use `jtsim -q`) and just check out the first few addresses of the sound CPU and compare them with MAME. Typically Z80 will start from address 0000 and very quickly jump somewhere else. If the beginning matches, the chances are that you got it right and you can move to test on the FPGA.

This is a rotated game, so it may be better to look at non rotated images to know which signals fail. If you think of the non-rotated one, graphics data seems ok vertically but half of the horizontal data for characters is wrong. Most likely the H/V components of the GFX memory address is not right. Also, there is some wrong tiles displayed on the sea, so the the tile address might be messed up with the H bit(s).

jotego, 2024-01-23:
I have taken the CPS cores to #511 as those are more complex. I will do those ones myself.

jotego, 2024-01-23:
A bit more information about the GFX:

GFX ROMs usually follow this format: `tile number, HVVV`, where the H represents one bit that toggles while reading the data to draw the image in a line (like left half of the tile vs right half). So H selects between two 4-pixel halves. The VVV represents three bits that tell you which line in an 8-pixel tall tile you are asking data for.

If tiles are 16x16, there is one more H and more V, like `tile number,HHVVVV`. Some games mix these bits in different ways: `HVVVHV`, etc. There is a small cache associated to each SDRAM slot in JTFRAME. For each to work well, data must be read in consecutive order. If a game uses the `HVVV` order, most cores will swap it to `VVVH`.  That swapping is done while writting the data to the SDRAM.

Most core swap the bits by identifying which ROM regions is being transferred to the SDRAM and shuffling the bits accordingly. Some more modern cores, like shouse, just tell in mem.yaml what they need. 

I hope this helps.

jotego, 2024-03-01:
completed in 31a39aeb

closed 2024-03-01 by jotego