jotego/jtcores #489: Odyssey: SEGA 315-0042 v1.0.1

jotego, 2024-01-13:
- [ ] devolver la versión en master a la fabricada
- [ ] arreglar la conexión de U16
- [ ] arreglar el símbol de U17 en la librería jt74
- [ ] revisar las conexiones VCC/VSS de los demás símbolos
- [ ] arreglar la PCB
- [ ] crear etiqueta en el esquemático para /VSYNC
- [ ] eliminar las anotaciones **0 Y 1**
- [ ] expandir la check list para verificar visualmente VCC y VSS en cada integrado
- [ ] pasar las reglas y la check list
- [ ] avanzar la versión a v1.0.2
- [ ] etiquetar la versión en git

etriana85, 2024-01-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

etriana85, 2024-01-18:
[image]

etriana85, 2024-01-18:
[image]
[image]

closed 2024-01-18 by etriana85