jotego/jtcores #493: contra,comsc: broken in 4c0d815ae745ce10cc552f8fbacc8f9d08714e48

jotego, 2024-01-18:
These two cores are not working in 4c0d815ae745ce10cc552f8fbacc8f9d08714e48. The problem must have happened much earlier.

gyurco, 2024-01-18:
Looks similar to the Karnov one...
Here (1001) the CPU should read "01", not "0F":
[image]

AFAIK RAM is in BRAM here, so no SDRAM controller is involved.

gyurco, 2024-01-18:
More possible that CEN rate again...the read of '1001' is only 2 master cycles. That would be a 24MHz 6809 (or 48?).

gyurco, 2024-01-18:
But CEN is usually good, only some reads are speed up...which makes the 'clock recovery' suspicious. I think my previous report about the flawed ram_ok is still stands, the input data is delayed by one cycle to ram_ok, so the CPU reads it too early.

gyurco, 2024-01-18:
Uhh, is it the intented CE?
It's similar to the driver who push full gas, then full break, and so on...
[image]

gyurco, 2024-01-18:
I don't thing cen3 delaying does anything - the cpu E and Q enable is generated in jtframe_sys6809.v using cen3 and rom_ok. Something must be done in that module if cen delaying is the way to go - but personally I think it would be ugly and could have many unwanted side-effects to other working cores, I would not touch it.

gyurco, 2024-01-18:
Additional to the above, CEN delaying doesn't work on RAM in BRAM access - as it doesn't have any OK signal, so if the clock recovery thinks it should run in full speed, the same problem occurs.

jotego, 2024-01-18:
Full speed recovery is still a 50% duty cycle, so the BRAM will work. That is by design.

Delay the rom_ok by one clock cycle. That should work. Then, we can keep registering cpu_din.

Also, note that *mx5k* may be failing too because cpu_din is also registered in that core and it shares the same _jtcontra_main_

gyurco, 2024-01-18:
_Full speed recovery is still a 50% duty cycle, so the BRAM will work._

...but...it doesn't :) Look at the first waveform. It's during the RAM test code, it fails very early.

I also noticed mx5k, the same fix can be applied. Meanwhile I discovered that the control rotation in Contra is flipped..maybe need to check dip_flip, too.

jotego, 2024-01-18:
If you run a simulation in the broken commit, verilator already stops and gives you a good hint:

```
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx1.u_attr_ram.u_ram cannot open file: gfx1_attr.bin
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx1.u_code_ram.u_ram cannot open file: gfx1_code.bin
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx1.u_obj_ram.u_ram cannot open file: gfx1_obj.bin
%Warning: gfx2_cfg.hex:0: $readmem file not found
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx2.u_attr_ram.u_ram cannot open file: gfx2_attr.bin
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx2.u_code_ram.u_ram cannot open file: gfx2_code.bin
WARNING: TOP.game_test.u_game.u_game.u_video.u_gfx2.u_obj_ram.u_ram cannot open file: gfx2_obj.bin
TOP.game_test.u_game.u_cen1_clk24 cen active while busy was high. Is busy dependent on a different clock domain?
- /home/jtejada/jtcores/modules/jtframe/hdl/inc/../../../../modules/jtframe/hdl/clocking/jtframe_gated_cen.v:84: Verilog $finish
```
This is related to both CPUs gating the cen signals, while they run at different rates. Undoing the cross gating in mem.yaml removes the warning:

```
clocks:
  clk24:
    - freq: 3000000
      gate: [ "main" ]
      outputs:
        - cen3
        - cen1p5
    - freq: 3579545
      gate: [  "snd" ]
      outputs:
        - cen_fm
        - cen_fm2
```
Still, when the cen recovery hits 50% the BRAM fails as you said. And the goal is to have both CPUs gating each other.

What this issue shows is that jtframe_sys6809, which hosts the BRAM, and jtframe_gated_cen are not that happy with each other. In the end one is a bit of a legacy scheme where the other one is the more modern approach.

If we just take the combinational logic, Verilator gives warnings and the jtframe_gated_cen also gives a critical warning and stops the sim. So I think we need to understand what the design limitation is better.

gyurco, 2024-01-18:
As the whole mc6809i is encapsulated by lint_off UNOTPFLAT, cpu_din is just a very little piece escaping from it.

About the other warning (cen active while busy was high. Is busy dependent on a different clock domain?) I don't know if is it a bug? Or just a real warning?

jotego, 2024-01-19:
> As the whole mc6809i is encapsulated by lint_off UNOTPFLAT, cpu_din is just a very little piece escaping from it.

That's right. We could just wrap that part with the same waiver. I have some memory of it failing more with the latest verilator version though. Anyway, I want to get rid of that mc6809i this year. So I would be ok to use that work around except for the other part of the problem:

> About the other warning (cen active while busy was high. Is busy dependent on a different clock domain?) I don't know if is it a bug? Or just a real warning?

This is the one that worries me. I'll check the file later.

One more aspect is this line:

`jtframe_sys6809 #(.RAM_AW(RAM_AW),.CENDIV(0))`

The `.CENDIV(0)` used to be 1, so the cen was divided by 4 internally. That would have prevented problems when registering the _din_ signals. But I was messing with something else, maybe **shouse** and Contra started running very slow. It may be running too fast now.

About the flip, see #232. MiST OSD should have the right orientation when you flip games. The value of the flip seen by the frame can be shown on the on-screen debug view too.

gyurco, 2024-01-19:
The OSD has the right flip, but the controls not...
As I see, core_mod[2] affects the OSD, but not the controls, which is strange. So I could hack to add bit[2] to core_mod (rom1=5 in MRA), and at the same time reversing dip_flip, too. It flipped the controls, and left the OSD at the same place.

This also makes me question where to put anticipated MRA changes? For example 1942 and variants need a 4 byte header starting from 32fa72a5165937e2694e8a42e49654c54d0f10af.

BTW, the CPU can be made to register the output. As a real 68xx puts data on the bus at the rising edge of E (to be able to read at the falling edge), dout can be registered using cen_Q for example (cen_E represents the falling edge of E). Of course users of dout should be aware that cpu_dout is valid at cen_E, not as soon as the address is seen on the bus.

gyurco, 2024-01-19:
_But I was messing with something else, maybe shouse and Contra started running very slow. It may be running too fast now._

It has the right speed with cen3 (3 MHz). With .CENDIV(1), it's way tooooo slow.

jotego, 2024-01-19:
> The OSD has the right flip, but the controls not... As I see, core_mod[2] affects the OSD, but not the controls, which is strange. So I could hack to add bit[2] to core_mod (rom1=5 in MRA), and at the same time reversing dip_flip, too. It flipped the controls, and left the OSD at the same place.

No hacks. Let's track the culprit. It looks like jtframe_inputs uses dip_flip to evaluate the control orientation. It should use **rotate** instead, which is generated in jtframe_dip and takes `core_mod[2]` into account. Change it to use that one instead. Change the input port name of jtframe_inputs accordingly too

> 
> This also makes me question where to put anticipated MRA changes? For example 1942 and variants need a 4 byte header starting from [32fa72a](https://github.com/jotego/jtcores/commit/32fa72a5165937e2694e8a42e49654c54d0f10af).

Sorry, I do not understand this question. 

> BTW, the CPU can be made to register the output. As a real 68xx puts data on the bus at the rising edge of E (to be able to read at the falling edge), dout can be registered using cen_Q for example [...]

I have not had problems with the output data, so I would not touch it for now.

> It has the right speed with cen3 (3 MHz). With .CENDIV(1), it's way tooooo slow (and has the strange clocking as I pasted on the second waveform).

What happens if you leave `.CENDIV(0)` but input 1.5MHz instead? I still think 3MHz with no division at all is not right for this CPU.

gyurco, 2024-01-19:
BTW, what I would suggest:
1. Make sure all cores compiles/run
2. Fix all the control/OSD rotation 
3. Never change the semantics of control/OSD rotation again after this

jotego, 2024-01-19:
>     1. Make sure all cores compile/run

Yes, that's issue #286 

>     2. Fix all the control/OSD rotation

Using rotate instead of dip_flip should work straight away in all cores. It should fix things for all of them because the OSD is the one thing that I check for all cores. I never use that rotate function. I do not know if anyone uses it. I do not offer it on MiSTer or Pocket which are the platforms with a substantial number of users.

>     3. Never change the semantics of control/OSD rotation again after this

Once things work, they tend not to change.

**MRA files**

MRA files are not source code so they are not in this repository but in JTBIN. Sometimes things in JTBIN lose consistency because of manual updates. I currently changing the workflow so I only do batch updates with all files together.

gyurco, 2024-01-19:
> Yes, that's issue #286
>
>
> Using rotate instead of dip_flip should work straight away in all cores. It should fix things for all of them because the OSD is the one thing that I check for all cores. I [...]
But if I use rotate, and it'll flip contra's controls, then it'll flip all others, too. Well, it's easier to rotate controls than rotating the display, at least for me :D

> > ```
> > 3. Never change the semantics of control/OSD rotation again after this
> > ```
> 
> Once things work, they tend not to change.
> 
+1

> **MRA files**
> 
> MRA files are not source code so they are not in this repository but in JTBIN. Sometimes things in JTBIN lose consistency because of manual updates. I currently changing the workflow so I only do batch updates with all files together.

I'm afraid that such changes (like add two bytes to header) will forgotten. If MRAs are not source code, then are they auto-generated from source code?

jotego, 2024-01-19:
The controls and the OSD should be rotated in the same way. Or am I missing something there? jtframe_inputs should rotate instead of dip_flip. It probably uses dip_flip because it has not received much attention over the years and has got out of sync with the rest of the code. 

Rotating the controls may not work for all games anyway. If it works for shooters, games like Commando, Gun Smoke... then that's enough. Nobody wants to play Contra on a non-rotated screen where the guy seems to be climbing! Whereas shooters are mostly playable on non-rotated screens.

**MRA**

MRA files are generated from the file `cfg/mame2mra.toml` when you type `jtframe mra contra`. All MRA and RBF files are normally transferred together to JTBIN. Only when I do things manually jumping over the scripts and check lists, do I end up with misplaced files.

jotego, 2024-01-19:
I would not mind dropping the rotated controls altogether, to be honest. But I think you use that, so we can keep it.

gyurco, 2024-01-19:
I've played Scramble a lot with rotated controls :)
I found it very useful in shooters, Commando and a lot of other games - maybe am I the only one?
Because on MiSTer they can rotate the image on the screen, on Pocket, they can rotate the device, maybe they're not missing it.

jotego, 2024-01-19:
We can close this one. Thanks for the PR. I have taken the jtframe_gated_cen assertion question to #497

closed 2024-01-19 by jotego